contiki-ng/contiki-ng #425: [Potential] Security Vulnerabilities within Erbium

bsmelo, 2018-04-10:
Hello developers of Contiki-ng/Erbium,

My name is Bruno, and I'm an MSc. student in Brazil within the Institute of Computing from the University of Campinas.
As part of my research on the application of fuzzing techniques for robustness and security black-box testing of CoAP implementations, I've tested your library. The sample used in my research was compiled from distribution/commit 9777ac4 @ 2018-02-03. The application used to test it was `examples/coap/plugtest-server.native`.

I'm contacting you because the application mentioned above was one of the samples for which our tool was able to detect robustness and/or security issues. In a broad sense, every failure we found can actually be classified as a security vulnerability, because they impact availability --- the application either aborts or needs forceful restart in order to restore servicing CoAP requests. However, we didn't go as far as performing a thorough root-cause analysis for those failures, since it would be unfeasible for us (more than 100 failures were detected across 25 samples, each one using a different CoAP library, spanning 8 programming languages) and thus out-of-scope of this particular research.

We think that one of our main contributions is the opportunity to make a real-world impact on IoT security by reporting those failures to CoAP libraries' maintainers, with a comprehensible and easy way to replicate them so developers can further investigate and fix those failures. So, in order to follow up with a responsible disclosure process, we ask for a proper e-mail address (or any other form of contact) so we can send you:
- A script to reproduce the failures;
- A pcap file used by the script, containing the packets causing the failures;
- A logfile with the stacktraces we got for each reported failure.

We expect a reply anytime soon.
Please let us know if which form of contact should we use --- or if it's ok to use this channel.

Thanks & Regards,
Bruno Melo.

simonduq, 2018-04-10:
Hi,
This is the right place to report such bugs. Please do share the script, pcap and logs, as well as complete instructions how to reproduce. We will investigate.
Thanks a lot for reporting!

bsmelo, 2018-04-10:
Hello,

We found 1 failure on this sample. You can find the script [1], the pcapng file [2] and the logfile [3] in this message.

The script uses Scapy [4] to read the packets from the pcapng file as well as sending them, which unfortunately requires sudo to send packets in the network. Although sudo for this is ugly (imho), you can still easily inspect the very small and simple script.

Please let us know if you were able to reproduce it and/or if further support is needed for that.

[1] http://www.students.ic.unicamp.br/~ra163122/653b8e286728cb777c81ba1e48df360b1a1a3d63b4ac06200ce35ce339f53c33.py
[2] http://www.students.ic.unicamp.br/~ra163122/653b8e286728cb777c81ba1e48df360b1a1a3d63b4ac06200ce35ce339f53c33.pcapng
[3] http://www.students.ic.unicamp.br/~ra163122/653b8e286728cb777c81ba1e48df360b1a1a3d63b4ac06200ce35ce339f53c33.log
[4] https://github.com/secdev/scapy/

nfi, 2018-04-12:
Thanks for reporting this and for the excellent test case to reproduce the problem! I made a PR #430 to address this issue.

bsmelo, 2018-04-12:
Great! =)

I see there are two CVEs reported for Contiki 3.0 at [1]. Should we apply for a CVE on this issue for Contiki-NG?
Also, I should probably report this to Contiki 3.0 team as well, since it looks like they could have the same problem, right?

[1] https://www.cvedetails.com/product/38087/Contiki-os-Contiki.html?vendor_id=16528

simonduq, 2018-04-12:
If the issue exists also in NG yes, please consider also reporting it!

g-oikonomou, 2018-05-04:
@nfi #430 closes this here, right?

nfi, 2018-05-04:
Yes, I think so.

closed 2018-05-04 by g-oikonomou