contiki-ng/contiki-ng #431: cc26xx-web-demo (LAUNCHXL-CC1310) cannot join 6lbr with Contiki-NG network

chenek, 2018-04-13:
After I build and run 6lbr with Contiki-NG on LAUNCHXL-CC1310, my cc26xx-web-demo (LAUNCHXL-CC1310) cannot join the network. LAUNCHXL-CC1310 running cc26xx-web-demo keeps fast blinking green LED and I also cannot see it pops on sensors tab of 6lbr web page. 
[image]

I also use sensniff to check over the air messages when I test this. In the screen shot, fe80::212:4b00:e00:cc00 is from 6lbr/Contiki-NG and fe80::212:4b00:a27:f319 is from LAUNCHXL-CC1310 running Contiki-NG cc26x-web-demo

[image]

Does anyone make contiki-ng cc26xx-web-demo joins 6lbr/Contiki-NG successfully? What do I miss?

simonduq, 2018-04-13:
Hi,

Did you check the limitations of 6lbr for NG? https://github.com/cetic/6lbr/wiki/Contiki-NG

Did you try with the border router included in NG? https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-RPL-border-router

chenek, 2018-04-16:
@simonduq Thanks for your help.
1. Yes, I had read https://github.com/cetic/6lbr/wiki/Contiki-NG.  I think if I build slip radio and cc26xx-web-demo for LAUNCHXL-CC1310, it should be no problem, isn't it? Do I need to take care of something when build those projects to run with cetic 6lbr/contik-ng?

2. When I tried to build border router included in NG on my Raspberry Pi, I get the following error. Do I miss something or do anything wrong? Is there any step-by-step document to show me how to run native border router in NG on Raspberry Pi? 

sudo ./border-router.native fd00::1/64
[INFO: Main      ] Starting Contiki-NG-develop/v4.1-25-gcb120e3
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: br-mac
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] Link-layer address 0102.0304.0506.0708
[INFO: Main      ] Tentative link-local IPv6 address fe80::302:304:506:708
[INFO: Native    ] Added global IPv6 address fd00::302:304:506:708
[INFO: RPL BR    ] Contiki-NG Border Router started
RPL-Border router started
border-router.native: can't open siodev: No such file or directory
../../os/services/rpl-border-router/native/Makefile.native:10: recipe for target 'connect-router' failed
make: *** [connect-router] Error 1
rm border-router.o
pi@raspberrypi:~/test/contiki-ng/examples/rpl-border-router $

simonduq, 2018-04-16:
Do you have a slip-radio device connected? You can specify a port with `-s` when running `./border-router.native`

chenek, 2018-04-16:
@simonduq Sorry, I am careless and don't build the native border router on Raspberry Pi. Can you show me how to build native border router on Raspberry Pi?

simonduq, 2018-04-16:
This should work on RPI just as well as on any Linux: https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-RPL-border-router#native-border-router

chenek, 2018-04-16:
@simonduq my question is what TARGET should I use when I do make on my Raspberry Pi?

simonduq, 2018-04-16:
`native`. But this is not the problem you're having: you're showing logs from a RPI, which suggest the binary is already compiled for the right `TARGET`.

chenek, 2018-04-16:
@simonduq  Thanks for the help. I see. Anyway, I just rebuild rpl0border-router and run sudo ./border-router.native fd00::1/64. However, I see the same issue. I am sure I have plugin LAUNCHXL-CC1310 running slip-radio to Raspberr Pi but it emulated as ttyACM0. So, the question is where to change siodev to ttyACM0?

simonduq, 2018-04-16:
https://github.com/contiki-ng/contiki-ng/issues/431#issuecomment-381505928

chenek, 2018-04-16:
@simonduq Now, rpl-broder-router seems running successfully on my Raspberry Pi. If I am correct, rpl-broder-router is not like 6lbr so I cannot access to [bbbb::100]. Is it correct? I can only try to join cc26xx-web-demo to rpl-broder-router and see if I can access to the web page on cc26xx-web-demo. Is this correct?

pi@raspberrypi:~/test/contiki-ng/examples/rpl-border-router $ sudo ./border-router.native fd00::1/64 -s /dev/ttyACM0
[INFO: Main      ] Starting Contiki-NG-develop/v4.1-25-gcb120e3
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: br-mac
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] Link-layer address 0102.0304.0506.0708
[INFO: Main      ] Tentative link-local IPv6 address fe80::302:304:506:708
[INFO: Native    ] Added global IPv6 address fd00::302:304:506:708
[INFO: RPL BR    ] Contiki-NG Border Router started
RPL-Border router started
********SLIP started on ``/dev/ttyACM0''
Opening tun interface:tun0
opened tun device ``/dev/tun0''
ifconfig tun0 inet `hostname` up
ifconfig tun0 add fd00::1/64
ifconfig tun0

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:127.0.1.1  P-t-P:127.0.1.1  Mask:255.255.255.255
          inet6 addr: fd00::1/64 Scope:Global
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

Setting prefix fd00::1
[INFO: BR        ] Server IPv6 addresses:
[INFO: BR        ]   fd00::212:4b00:e00:cc00
[INFO: BR        ]   fe80::212:4b00:e00:cc00

simonduq, 2018-04-16:
At the risk of repeating myself, this is covered in https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-RPL-border-router

simonduq, 2018-04-17:
The native BR runs the shell, you should be able to interact from stdin/stdout. Try to print RPL status, routes and neighbors from it. Type `help` for a list of commands.

chenek, 2018-04-17:
@simonduq Thanks a lot. I do "ip-nbr" in the shell and I can see my cc26xx-web-demo is listed by it. According to my current test, cc26xx-web-demo can join rpl-border-router but why cc26xx-web-demo cannot join 6lbr/contiki-NG?

simonduq, 2018-04-17:
Probably because you're on RPL-Lite (Contiki-NG's default, and most solid option) whereas 6lbr currently only supports RPL-Classic and storing mode.
https://github.com/cetic/6lbr/wiki/Contiki-NG

simonduq, 2018-04-17:
Please really consider browsing our wiki, we now have a lot of doc over there.
For configuration https://github.com/contiki-ng/contiki-ng/wiki/The-Contiki%E2%80%90NG-configuration-system

chenek, 2018-04-18:
@simonduq I had read the wiki docs but not aware how to do it. Now I get it. However, I get the following compile error when I do "make TARGET=srf06-cc26xx BOARD=launchpad/cc1310 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC cc26xx-web-demo.bin" to build cc26xx-web-demo.

  CC        ../../../../arch/cpu/cc26xx-cc13xx/lib/cc13xxware/startup_files/ccfg.c
  CC        ../../../../arch/cpu/cc26xx-cc13xx/./ieee-addr.c
  CC        ../../../../arch/cpu/cc26xx-cc13xx/./fault-handlers.c
  CC        ../../../../arch/cpu/cc26xx-cc13xx/lib/cc13xxware/startup_files/startup_gcc.c
  CC        cc26xx-web-demo.c
  CC        cetic-6lbr-client.c
cetic-6lbr-client.c: In function 'timeout_handler':
cetic-6lbr-client.c:162:7: error: implicit declaration of function 'rpl_get_default_instance' [-Werror=implicit-function-declaration]
       instance = rpl_get_default_instance();
       ^
cetic-6lbr-client.c:162:16: error: assignment makes pointer from integer without a cast [-Werror]
       instance = rpl_get_default_instance();
                ^
cetic-6lbr-client.c:163:30: error: 'rpl_instance_t' has no member named 'dag'
       if(instance && instance->dag.preferred_parent) {
                              ^
cetic-6lbr-client.c:164:59: error: 'rpl_instance_t' has no member named 'dag'
         add_ipaddr(buf + i, rpl_parent_get_ipaddr(instance->dag.preferred_parent));
                                                           ^
cc1.exe: all warnings being treated as errors
make: *** [../../../../Makefile.include:317: obj_srf06-cc26xx/cetic-6lbr-client.o] Error 1
rm obj_srf06-cc26xx/fault-handlers.o obj_srf06-cc26xx/startup_gcc.o cc26xx-web-demo.o

Is this a bug in Contiki-NG? How can I fix it?

simonduq, 2018-04-18:
Looks like you didn't clean your build

chenek, 2018-04-18:
@simonduq I git the latest version of https://github.com/contiki-ng/contiki-ng today. I just do "make clean" and then "make TARGET=srf06-cc26xx BOARD=launchpad/cc1310 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC cc26xx-web-demo.bin". However, I see the same problem. Can you check if there is something wrong in current source code?

simonduq, 2018-04-18:
Specify the `TARGET` when cleaning, or else use `distclean`: https://github.com/contiki-ng/contiki-ng/wiki/The-Contiki%E2%80%90NG-build-system

chenek, 2018-04-18:
@simonduq Just try both "make TARGET=srf06-cc26xx clean" and "make distclean" before I do "make TARGET=srf06-cc26xx BOARD=launchpad/cc1310 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC cc26xx-web-demo.bin". However, I still get the same compile error.

simonduq, 2018-04-18:
Oh sorry, I was trying with the wrong example.
It does look like a bug on our `cetic-6lbr-client.c`, yes.
Only supports RPL Lite but not RPL Classic.

Added a new issue: https://github.com/contiki-ng/contiki-ng/issues/442

chenek, 2018-04-18:
Thanks for verifying and confirm this issue.

chenek, 2018-04-19:
@simonduq  and @g-oikonomou I see #443 . I test it and confirm it can work with 6lbr/Contik-NG now. You are great!

closed 2018-04-19 by chenek

simonduq, 2018-04-19:
Thanks @chenek for reporting the issue and confirming the fix!