contiki-ng/contiki-ng #435: DS18B20 code

marcomilazzo, 2018-04-14:
Hi
a sample code for the sd18b20 in cc1310/ cc2650.

/* It works without the 4.7K resistor between VCC and the in pin of the sensor */
/* includ the math.h for printing floats */

#include "contiki.h"
#include "ti-lib.h"
#include "ioc.h"
#include <stdio.h>
#include <string.h>
#include <math.h>
 
#define DS18B20_COMMAND_READ_SCRATCH_PAD 0xBE
#define DS18B20_COMMAND_START_CONVERSION 0x44
#define DS18B20_COMMAND_SKIP_ROM 0xCC
#define DS18B20_PORT 1     // assign the port you want to use

/*---------------------------------------------------------------------------*/
PROCESS(ds18b20, "ds18b20");
AUTOSTART_PROCESSES(&ds18b20);
/*---------------------------------------------------------------------------*/

/*-------------------------------------------------------------------------*/
void OW_SET_OUTPUT(){
  ti_lib_rom_ioc_pin_type_gpio_output(DS18B20_PORT);
  ti_lib_gpio_write_dio(DS18B20_PORT, 0);
}
/*-------------------------------------------------------------------------*/
/*-------------------------------------------------------------------------*/
void OW_SET_INPUT(){
  ti_lib_rom_ioc_pin_type_gpio_input(DS18B20_PORT);
  IOCIOPortPullSet(DS18B20_PORT,IOC_IOPULL_UP );  //IOC_IOPULL_UP  IOC_NO_IOPULL  IOC_IOPULL_DOWN
 }
/*-------------------------------------------------------------------------*/

/*-------------------------------------------------------------------------*/
/* probe_for_ds18b20 probes for the sensor. Returns 0 on failure, 1 on success
 * Assumptions: only one sensor on the "1-wire bus", on port WSN_DS18B20_PORT
 * BUG: THIS CODE DOES NOT WORK AS INTENDED! IT RETURNS "1" EVEN WHEN THERE
 * IS NO SENSOR CONNECTED.*/
 uint8_t ds18b20_probe(void)
{
  uint8_t result = 0;
 
  /* Reset 1W-bus */
  /* Pull PIN low for 480 microseconds (us)
   * Start with setting bit DS18B20_1_PIN to 0 */
  /* then set direction to OUT by setting DS18B20_1_DDR bit to 1 */
  OW_SET_OUTPUT();
  /* Delay 480 us */
  clock_delay_usec(480);

  /* See if sensor responds. First release the bus and switch to INput mode
   * by setting DS18B20_1_DDR bit to 0 */
  OW_SET_INPUT();
  /* Activate internal pull-up by  setting pin to HIGH (when in INput mode)
   * Wait for the pin to go HIGH for 64 us */
  clock_delay_usec(64);

  /* Now the sensor, if present, pulls the pin LOW for 60-240 us
   * Detect 0 on PIND bit DS18B20_1_PIN. Invert the result so a presence
   * (aka * a 0)  sets "result" to 1 (for success) */
  result = !GPIO_readDio(DS18B20_PORT); //OW_GET_PIN_STATE()

  /* The sensor releases the pin so it goes HIGH after 240 us, add some
     for the signal to stabilize, say 300 usecs to be on the safe side? */
  if(result) {
    clock_delay_usec(300);
    /* Now the bus should be HIGH again */
[... 223 lines cut ...]
 
  while(1) {
		etimer_set(&periodic, CLOCK_SECOND * 3);

	
    	ret = ds18b20_probe();
		if (ret== 0) {  printf("Error Resetting\n"); }
		
		ret= ds18b20_get_temp(&temp );
		//if (ret== 1) {  printf("Temp= %ld.%03u\n" , (long)temp,(unsigned)((temp-floor(temp))*1000)); } //3 digits
		if (ret== 1) {  printf("Temp= %ld.%-1.2u\n" , (long)temp,(unsigned)((temp-floor(temp))*100)); } //2 digits
		if (ret== 0) {  printf("Error Reading\n"); }

		//double num =temp;
		//int intpart = (int)num;
		//double decpart = num - intpart;
		//printf("num1 = %d.%u\n" , intpart,(unsigned)((decpart-floor(decpart))*1000));

	
	
	   PROCESS_WAIT_UNTIL(etimer_expired(&periodic));
  }
  PROCESS_END();
}
/*---------------------------------------------------------------------------*/

g-oikonomou, 2018-04-16:
@marcomilazzo having those as pasted text in an issue makes things very difficult to review and manage. If you wish this to be considered, can you please convert it to a proper pull request and also format the code properly? The build system may need adjusted too, we'll see.

I'll be closing this one.

closed 2018-04-16 by g-oikonomou

marcomilazzo, 2018-04-16:
hi
i tried to open a pullwith no sucess !the code is for you and use ad yoi like
i will try again