contiki-ng/contiki-ng #436: Pwm sample code for cc1310/cc2650

marcomilazzo, 2018-04-14:
Hi
this code is a modified version of buzzer.c of the mqtt demo .it has controll of the duty cycle
ciao

#include "contiki.h"
#include "ti-lib.h"
#include "lpm.h"
#include <stdio.h>
#include <string.h>

#define PWM_PIN 1 

LPM_MODULE(pwm_module, NULL, NULL, NULL, LPM_DOMAIN_PERIPH);

/*---------------------------------------------------------------------------*/
PROCESS(pwm, "pwm");
AUTOSTART_PROCESSES(&pwm);
/*---------------------------------------------------------------------------*/

/*---------------------------------------------------------------------------*/
void pwm_start(int freq,int duty)
{
  uint32_t load;

  /* Enable GPT0 clocks under active, sleep, deep sleep */
  ti_lib_prcm_peripheral_run_enable(PRCM_PERIPH_TIMER0);
  ti_lib_prcm_peripheral_sleep_enable(PRCM_PERIPH_TIMER0);
  ti_lib_prcm_peripheral_deep_sleep_enable(PRCM_PERIPH_TIMER0);
  ti_lib_prcm_load_set();
  while(!ti_lib_prcm_load_get());

  /* Drive the I/O ID with GPT0 / Timer A */
  ti_lib_ioc_port_configure_set(PWM_PIN, IOC_PORT_MCU_PORT_EVENT0,
                                IOC_STD_OUTPUT);

  /* GPT0 / Timer A: PWM, Interrupt Enable */
  HWREG(GPT0_BASE + GPT_O_TAMR) = (TIMER_CFG_A_PWM & 0xFF) | GPT_TAMR_TAPWMIE;
  /*
   * Register ourself with LPM. This will keep the PERIPH PD powered on
   * during deep sleep, allowing the pwm to keep working while the chip is
   * being power-cycled
   */
  lpm_register_module(&pwm_module);

  /* Stop the timer */
  ti_lib_timer_disable(GPT0_BASE, TIMER_A);

  if(freq > 0) {
    load = (GET_MCU_CLOCK / freq);
    ti_lib_timer_load_set(GPT0_BASE, TIMER_A, load);
    ti_lib_timer_match_set(GPT0_BASE, TIMER_A,duty );

    /* Start */
    ti_lib_timer_enable(GPT0_BASE, TIMER_A);
  }
}
/*---------------------------------------------------------------------------*/

/*---------------------------------------------------------------------------*/
void pwm_stop()
{
  /*
   * Unregister the pwm module from LPM. This will effectively release our
   * lock for the PERIPH PD allowing it to be powered down (unless some other
   * module keeps it on)
   */
  lpm_unregister_module(&pwm_module);

  /* Stop the timer */
  ti_lib_timer_disable(GPT0_BASE, TIMER_A);

  /*
   * Stop the module clock:
   *
   * Currently GPT0 is in use by clock_delay_usec (GPT0/TB) and by this
   * module here (GPT0/TA).
   *
   * clock_delay_usec
   * - is definitely not running when we enter here and
   * - handles the module clock internally
   *
   * Thus, we can safely change the state of module clocks here.
   */
  ti_lib_prcm_peripheral_run_disable(PRCM_PERIPH_TIMER0);
  ti_lib_prcm_peripheral_sleep_disable(PRCM_PERIPH_TIMER0);
  ti_lib_prcm_peripheral_deep_sleep_disable(PRCM_PERIPH_TIMER0);
  ti_lib_prcm_load_set();
  while(!ti_lib_prcm_load_get());

  /* Un-configure the pin */
  ti_lib_ioc_pin_type_gpio_input(PWM_PIN);
  ti_lib_ioc_io_input_set(PWM_PIN, IOC_INPUT_DISABLE);
}
/*---------------------------------------------------------------------------*/

/*---------------------------------------------------------------------------*/
int duty_cycle_step(int freqHz,int total_duty_steps ){
  int load;  int duty_step;
  
  load = (GET_MCU_CLOCK / freqHz);
  duty_step = load /total_duty_steps;   
 return duty_step;
}
/*---------------------------------------------------------------------------*/

/*---------------------------------------------------------------------------*/
void demo_dimming_0_100(int freqHz,int total_duty_steps, int incremet_duty_step){
 static int count;  int duty; int duty_step;
 
 duty_step= duty_cycle_step(freqHz,total_duty_steps);  
 
 if (count  >= total_duty_steps) { count=-1; }
 count = count + incremet_duty_step ;
 duty = count * duty_step;
   
 pwm_start(freqHz,duty);                     
 printf("Duty_Cycle:%d/%d  Value:%d\n",count,total_duty_steps ,duty   );
}
/*---------------------------------------------------------------------------*/

/*---------------------------------------------------------------------------*/
void dimming_0_100(int freqHz,int total_duty_steps, int duty){
 int calc_duty; int duty_step;
 
 duty_step= duty_cycle_step(freqHz,total_duty_steps);  
 calc_duty = duty * duty_step;
 pwm_start(freqHz,calc_duty);                     
 
 printf("Duty_Cycle:%d/%d  Value:%d\n",duty ,total_duty_steps,calc_duty   );
}
/*---------------------------------------------------------------------------*/

/*---------------------------------------------------------------------------*/
PROCESS_THREAD(pwm, ev, data)
{
  static struct etimer periodic;
   
  PROCESS_BEGIN();
  
        printf("Contiki CC26xx/CC1310 PWM demo using DIO1 \r\n");
 	
  while(1) {
        etimer_set(&periodic, CLOCK_SECOND * 1);
        
	  
	    //demo_dimming_0_100(1000,10,1); // frequebcy(period) ,wanted steps, increment
	    
		dimming_0_100(1000,10,3); // frequebcy(period) ,wanted steps, duty
	  
    	PROCESS_WAIT_UNTIL(etimer_expired(&periodic));
  }

  PROCESS_END();
}
/*---------------------------------------------------------------------------*/

g-oikonomou, 2018-04-16:
Same comment as in #435, this needs converted to a pull request if possible.

closed 2018-04-16 by g-oikonomou

marcomilazzo, 2018-04-16:
I'm sorry
can yoi explain me how and were to do it?
marco