contiki-ng/contiki-ng #440: Failure to run cc26xx-web-demo MQTT publish to IBM Watson Iot Platform.

chenek, 2018-04-17:
I try to to run cc26xx-web-demo MQTT publish to IBM Watson Iot Platform. I create the following device on  IBM Watson Iot Platform.

[image]

I get Device Type : cc26xx-ibm-iot, Organization ID : uc6bmi and Authentication Token (I don't show here). I fill those information into cc26xx-web-demo MQTT/IBM cloud config page.

[image]

After submit and MQTT reconnect, it always shows not authorized on IBM Watson Iot Platform.

[image]

I have a feeling that I might use wrong Type ID in cc26xx-web-demo MQTT/IBM cloud config page. Since I don's see Type ID on IBM Watson Iot Platform when I create device and I only see Device Type, I use my Device Type: cc26xx-ibm-iot to fill Type ID in cc26xx-web-demo MQTT/IBM cloud config page. Is this the reason why I get not authorized feedback? If so, how can I get correct Type ID?

g-oikonomou, 2018-04-17:
I've not used that in a while, so I am reading the docs again ([This page](https://console.bluemix.net/docs/services/IoT/reference/security/connect_devices_apps_gw.html#connect_devices_apps_gw) and [this page](https://console.bluemix.net/docs/services/IoT/devices/mqtt.html#mqtt)).

For devices, the client ID should be in the format `d:orgId:deviceType:deviceId` and as far as I can see in the code we are doing that. `deviceType` corresponds to our Type ID, so I cannot immediately see anything wrong with what you are trying. Did you try quickstart without auth?

chenek, 2018-04-17:
Yes, QuickStart works fine. When I test with Org ID and Auth token from IBM Watson IOT Platform, what IPv6 address should I use? Do I have to change it from QuickStart IPv6 address?

g-oikonomou, 2018-04-17:
Regarding IP addresses, I'm pretty sure you didn't have to in the past. If you are getting those errors it means your device is likely talking to the platform correcty. But if I can understand the docs correctly maybe you need to use  the IP address of `uc6bmi.messaging.internetofthings.ibmcloud.com`. What IP address are you using? Where did you get it from?

From what I can see again in the docs, username should always be `use-token-auth` and then the password is the token that you will have generated when you registered the device with the platform. Are the Device IDs the same on both sides?

g-oikonomou, 2018-04-17:
It would be very useful if you could run wireshark on some interface where the traffic goes through and then capture the MQTT connect message and the MQTT publish message (if one ever goes out)

chenek, 2018-04-18:
In previous test, I use 0064:ffd9:0000:0000:0000:0000:b8ac:7cbd which is QuickStart IPv6 address as Broker IP. Today, I try to use 0064:ffd9:0000:0000:0000:0000:A92D:0214 (IPv4 address of uc6bmi.messaging.internetofthings.ibmcloud.com is 169.45.2.20 and IPv6 is 0064:ffd9:0000:0000:0000:0000:A92D:0214) as Broker IP to test again and it still doesn't work. I attach the wireshark log

[uc6bmi.messaging.internetofthings.ibmcloud.com.pcapng.zip](https://github.com/contiki-ng/contiki-ng/files/1922233/uc6bmi.messaging.internetofthings.ibmcloud.com.pcapng.zip)

chenek, 2018-04-18:
@g-oikonomou I also use 0064:ffd9:0000:0000:0000:0000:b8ac:7cbd which is QuickStart IPv6 address as Broker IP to test agin and it still doesn't work. The following attachment is wireshark sniffer log.

[ibmquickstart.pcapng.zip](https://github.com/contiki-ng/contiki-ng/files/1922511/ibmquickstart.pcapng.zip)

chenek, 2018-04-18:
@g-oikonomou When I fill those information into cc26xx-web-demo MQTT/IBM cloud config page and click submit, I see "Auth Token" turn to blank but other fields not. Is this a problem?

[image]

g-oikonomou, 2018-04-19:
130 is indeed `MQTT_EVENT_CONNECTION_REFUSED_ERROR`. No idea what might be causing this, it has worked fine in the past. Pretty certain by now it must be related to the authentication.

It seems you _are_ receiving the `CONNACK` message correctly. I need to look inside it, it typically carries the reason for connection failure.

g-oikonomou, 2018-04-19:
Of the two pcaps that you have linked above, which is the one that did not work? I can see successful MQTT connection establishment in both of them

chenek, 2018-04-19:
@g-oikonomou I do the wireshark sniffer again and attach it for your checking.
[MQTT_EVENT_CONNECTION_REFUSED_ERROR.pcapng.zip](https://github.com/contiki-ng/contiki-ng/files/1927325/MQTT_EVENT_CONNECTION_REFUSED_ERROR.pcapng.zip)

BlueberryTech01, 2018-04-19:
Hi all, I'm having the same problem of @chenek . I tried translating my IPv4 of the cloud into IPv6 but nothing. How can I enter the auth-token?

BlueberryTech01, 2018-04-19:
[image]

I see this. I using the ipv6 of my address orgid.messaging.internetofthings.ibmcloud.com. Problem with auth-token?

g-oikonomou, 2018-04-19:
These PCAPs are very helpful.

The the last PCAP, frame 31 is the `CONNECT` request, which is responded to with a `CONNACK`. This is what I expected to see. In the `CONNACK` the server returns response code 0x05: "Connection Refused: not authorized" and then immediately terminates the TCP connection.

In the `CONNECT` message, I was expecting to see the username bit set, the password bit set and then the username `use-token-auth` and your token in the payload. This is what the code tries to do, but I am not seeing those so something smells. Let me investigate a bit more. perhaps I remember something incorrectly.

g-oikonomou, 2018-04-19:
By the way, the Auth Token field of the web page will always be blank after the page loads, this is intentional.

BlueberryTech01, 2018-04-19:
@g-oikonomou for adding the auth-token? Do you seen the my log message?

g-oikonomou, 2018-04-19:
@BlueberryTech01 the main concern I have about the log you posted is why it says `userID()`, it seems to be blank. I am not sure what it should say there, but I would have guessed it should say `use-token-auth`. I need to look at the IBM docs from closer and also need to verify what our client actually tries to do. I'll need to look into that.

Also, my mind could be playing tricks here, but I think I have read somewhere in the past the the platform will only accept token auth over SSL/TLS. I don't remember where I read this and I cannot find it now. Could be the reason, so if anyone spots anything in the docs...

BlueberryTech01, 2018-04-19:
This @g-oikonomou https://console.bluemix.net/docs/services/IoT/reference/security/connect_devices_apps_gw.html#connect_devices_apps_gw .

I can not connect without auth?

g-oikonomou, 2018-04-19:
Heya @chenek @BlueberryTech01 

I ran some tests in my rig: SmartRF06/cc2650EM web demo + zoul/firefly border router. I am running mosquitto broker on my localhost (device where the BR is connected).

I built the example as-is and let it run. In the attached PCAP, see how the device sends `CONNECT` (frame 96) with Client ID `d:quickstart:cc26xx:00124b8b858c`. In the `CONNECT` message, the username flag and password flag are both clear. This is as expected and the `CONNACK` (frame 98) returns with code 0x00 "Connection Accepted". This is the mosquitto trace:

```
1524172345: New connection from fd00::212:4b00:68b:858c on port 1883.
1524172345: New client connected from fd00::212:4b00:68b:858c as d:quickstart:cc26xx:00124b8b858c (c1, k11520).
1524172345: Sending CONNACK to d:quickstart:cc26xx:00124b8b858c (0)
1524172346: Received PUBLISH from d:quickstart:cc26xx:00124b8b858c (d0, q0, r0, m0, 'iot-2/evt/status/fmt/json', ... (182 bytes))
1524172376: Received PUBLISH from d:quickstart:cc26xx:00124b8b858c (d0, q0, r0, m0, 'iot-2/evt/status/fmt/json', ... (176 bytes))
```

I then used the web interface to change Org ID to "something" and set a token. See screenshot.

[image]

Hit submit and, as intended, the MQTT web demo client disconnected and tried to reconnect to the broker with the new params. The `CONNECT` message (frame 294 in the PCAP) header now has both username and password flags set. In the message payload, there exists the username (`use-token-auth`), as I wrote above that I anticipated) and also the password. The broker responds with `CONNACK` (frame 296), code 0x00 "Connection Accepted". This is the mosquitto trace:

```
1524172382: Received DISCONNECT from d:quickstart:cc26xx:00124b8b858c
1524172402: New connection from fd00::212:4b00:68b:858c on port 1883.
1524172402: New client connected from fd00::212:4b00:68b:858c as d:something:cc26xx:00124b8b858c (c1, k11520, uuse-token-auth).
1524172402: Sending CONNACK to d:something:cc26xx:00124b8b858c (0)
1524172402: Received SUBSCRIBE from d:something:cc26xx:00124b8b858c
1524172402: 	iot-2/cmd/+/fmt/json (QoS 0)
1524172402: d:something:cc26xx:00124b8b858c 0 iot-2/cmd/+/fmt/json
1524172402: Sending SUBACK to d:something:cc26xx:00124b8b858c
```

Note how the Client ID changes from `d:quickstart:...` to `d:something:...`

So:
* The client is doing what I am expecting it to be doing
* In the erroneous PCAP that @chenek posted above ( MQTT_EVENT_CONNECTION_REFUSED_ERROR.pcapng.zip), I am not seeing what I am expecting to see: The client is not sending the correct credentials, username and password bits in the header are clear and the payload does not contain the credentials. So we are missing something here

[mqtt-tests.pcapng.zip](https://github.com/contiki-ng/contiki-ng/files/1929835/mqtt-tests.pcapng.zip)

BlueberryTech01, 2018-04-19:
Hi @g-oikonomou,

Can you need the Auth-token to access the cloud? From how I read in IBM's documents,the port 1883 doesn't need security. It's possible?

g-oikonomou, 2018-04-19:
The IBM cloud has two flavours:

* Quickstart: No auth required
* Registered devices (many more features): You need token authentication

Port 1883 is MQTT unencrypted (as opposed to over SSL/TLS)

chenek, 2018-04-20:
@g-oikonomou In previous wireshark log, I hard code Type ID, Org ID, and Auth Token into cc26xx-web-demo.h and mqtt-client.c. I believe that's the reason why you don't see the username bit set and the password bit set. By the way, I don't know where to check this username bit and password bit in wireshark. Can you do a screenshot to show me how to check it? 

This morning, I try to fill Type ID, Org ID, and Auth Token into MQTT/IBM Cloud Config of cc26xx-web-demo instead hard code it to test again (I also enable debug log output to UART) and I get some progress. The followings are the UART output from cc26xx-web-demo and you can see it shows "APP - MQTT Disconnect. Reason 1". I also attach the wireshark log of this test for you to check.

[MQTT DISCONNECT REASON 1.pcapng.zip](https://github.com/contiki-ng/contiki-ng/files/1930416/MQTT.DISCONNECT.REASON.1.pcapng.zip)

BlueberryTech01, 2018-04-20:
@g-oikonomou Then, I cannot to communicate with cloud? Maybe that the Contiki OS don't support the TLS protocoll. Is correct?

g-oikonomou, 2018-04-20:
@BlueberryTech01 that is correct, we do not have MQTT over TLS at the moment.

@chenek this PCAP makes a lot more sense.

> In previous wireshark log, I hard code Type ID, Org ID, and Auth Token into cc26xx-web-demo.h and mqtt-client.c. I believe that's the reason why you don't see the username bit set and the password bit set.

The logic is: If Type ID is quickstart, the client will try to connect without auth. If Type ID is anything else, it will try to connect using `use-token-auth` and a token. The same logic should be followed by the client if you hardcode a different Client ID as well as if you change from quickstart to something else (or vice versa) over the web interface. If the two behaviours are different then we are looking at a bug. So in theory hard-coding a different Type ID and token should not break things and you should get a good PCAP.

> I don't know where to check this username bit and password bit in wireshark. Can you do a screenshot to show me how to check it?

See the attached screenshot.

[image]

But my concern is this sequence of events:

```
APP - Subscribing!
[...]
MQTT - Got mqtt_do_subscribe_mqtt_event!
MQTT - Sending subscribe message! topic iot-2/cmd/+/fmt/json topic_length 20
[...]
MQTT - Got TCP_DATA_SENT
MQTT - Disconnected by tcp event 1
APP - MQTT Disconnect. Reason 1
Disconnected
Disconnected. Attempt 2 in 1024 ticks
```

As you can see, the TCP connection drops immediately after the subscribe request is sent. This is confirmed by the PCAP: Frame 29 is the subscribe request, which is correctly ACKed (by TCP) in frame 31. I would then expect the server to respond with a `SUBACK`, but instead of that it terminates the TCP connection, starting with frame 34. This is a graceful TCP connection termination. I am not sure why the IBM platform does not like this subscribe attempt.

One way to debug this is by removing the subscribe attempt in L783 `subscribe();`. This will allow you to see if authenticated publishing works.

g-oikonomou, 2018-04-20:
I wrote Type ID where I should have written Org ID. Edited

BlueberryTech01, 2018-04-20:
@g-oikonomou So how do I communicate with the cloud? How are you trying?

chenek, 2018-04-20:
@g-oikonomou May I know what wireshark version you use? In you screen shot, I see you fill mqtt in top left edit box and you can filter all MQTT  messages but I don't see it on my wireshark. I do comment out "subscribe();" to test with publishing only today but the result is the same. I don't do wireshark log and also don't download UART output when I test this. I can redo this later and post again. By the way, can you make your kit to connect to IBM Watson Iot platform?

chenek, 2018-04-20:
@g-oikonomou I install latest wireshark version and I can see similar in your screen shot.

chenek, 2018-04-20:
@g-oikonomou  I find I can see similar result in your screen shot with wireshark 2.4.6 to open your log in mqtt-tests.pcapng.zip. However, if I use the same version to open my log like MQTT_EVENT_CONNECTION_REFUSED_ERROR.pcapng.zip, I cannot see MQTT in protocol field and I only see 802.15.4. How do you decrypt my wireshark log? Is this related to my wireshark version when I run sensniff? Sorry to bother you again and again.

g-oikonomou, 2018-04-20:
I am using WS 2.4.2 on Mac OS X. This version has opened all PCAPs posted in this thread and parsed frames no problem at all

g-oikonomou, 2018-04-20:
> By the way, can you make your kit to connect to IBM Watson Iot platform?

I have not used NAT64 in quite a while now, so currently no, I am not in a position to easily get nodes to talk to the IBM cloud

BlueberryTech01, 2018-04-20:
@g-oikonomou If I want to communicate with cloud for to save the data. How can I do?

g-oikonomou, 2018-04-21:
Sorry @chenek I am not 100% sure what IP address should be used. When this example was developed we were just using the same IP addr for quickstart as well as for registered devices, I don't remember every having to change it. Using uc6bmi.messaging.internetofthings.ibmcloud.com  makes sense.

I cannot guess why the platform is refusing the connection.

chenek, 2018-04-21:
@g-oikonomou any comment about how to keep debugging to fix this issue?

g-oikonomou, 2018-04-21:
@chenek 
> any comment about how to keep debugging to fix this issue?

I would debug without the example. I would register a device the way you have done. I would then use  `mosquitto_pub` or other MQTT client to try and understand / reverse engineer the situation. I would try connecting over unencrypted TCP vs over TLS.

For mosquitto, I would start with something like:

```
mosquitto_pub -h uc6bmi.messaging.internetofthings.ibmcloud.com -u use-token-auth -P <your token> -i d:uc6bmi:cc13xx:00124b27f319 -t iot-2/evt/status/fmt/json -m "{"d":{"myName":"TI SmartRF06EB + CC13xx EM","Seq #":1,"Uptime (sec)":34,"Def Route":"fe80::212:4b00:e00:cc00","RSSI (dBm)":-48,"Battery Temp (C)":22,"Battery Volt (mV)":3292}}"
```

I would be running an MQTT packet capture on your device's outgoing interface to see what happens.

Once confident I knew how to connect using username/token, I'd try to replicate using an actual device.

I am now strongly starting to suspect the IBM platform refuses authenticated connections over unsecure channel:

> If you want to connect your device without registering it, for example to run a proof of concept, you can do so by using the special organization ID QuickStart. QuickStart is a public sandbox instance of Watson IoT Platform that runs in the cloud. If you do not need to connect securely, you can use QuickStart to test your device connectivity and experiment with Watson IoT Platform. When you have finished experimenting, reconnect your device securely to your own specific organization ID instance by using TLS and your authentication token.

https://console.bluemix.net/docs/services/IoT/devices/device_dev_index.html#device_dev_index

chenek, 2018-04-21:
@g-oikonomou 

> I am now strongly starting to suspect the IBM platform refuses authenticated connections over unsecure channel:

I suspected the same thing and will do the same test to see if that is true.

g-oikonomou, 2018-04-21:
You may benefit by starting with publishing to quickstart using `mosquitto_pub` until you get comfortable with the CLI options. Let us know how it goes!

chenek, 2018-04-21:
@g-oikonomou I am comfortable with CLI and will test this first. Will update here once I have any finding. Thanks for your help.

chenek, 2018-04-22:
@g-oikonomou I just test mosquitto_pub as the following command but I get "Connection Refused: not authorised."

pi@raspberrypi:~/ng/6lbr/examples/6lbr $ mosquitto_pub -h uc6bmi.messaging.internetofthings.ibmcloud.com -u use-token-auth -P Ti23570163 -i d:uc6bmi:cc13xx:00124b27f319 -t iot-2/evt/status/fmt/json -m "{"d":{"myName":"LAUNCHXL-CC1310","Seq #":1,"Uptime (sec)":34,"Def Route":"fe80::212:4b00:e00:cc00","RSSI (dBm)":-48,"Battery Temp (C)":22,"Battery Volt (mV)""
Connection Refused: not authorised.
Error: The connection was refused.

I also try "--tls-version" in mosquitto_pub to use TLS as the following command but I still get Connection Refused: not authorised.

pi@raspberrypi:~/ng/6lbr/examples/6lbr $ mosquitto_pub -h uc6bmi.messaging.internetofthings.ibmcloud.com -u use-token-auth -P Ti23570163 -i d:uc6bmi:cc13xx:00124b27f319 -t iot-2/evt/status/fmt/json -m "{"d":{"myName":"LAUNCHXL-CC1310","Seq #":1,"Uptime (sec)":34,"Def Route":"fe80::212:4b00:e00:cc00","RSSI (dBm)":-48,"Battery Temp (C)":22,"Battery Volt (mV)"" --tls-version
Connection Refused: not authorised.
Error: The connection was refused.

What's your opinion?

chenek, 2018-04-22:
@g-oikonomou after I suf some IBM documents, I read this one https://console.bluemix.net/docs/services/IoT/reference/security/connect_devices_apps_gw.html#connect_devices_apps_gw and figure that IBM Watson IoT service uses TLS as default. After I set it to TLS optional, it works flawlessly. Thanks for your great support.

g-oikonomou, 2018-04-22:
@chenek glad you got it to work. Can you please explain in a little more detail what you mean "set it to TLS optional" and how you did it? It'd be nice to document that in the respective example READMEs

chenek, 2018-04-23:
@g-oikonomou I create pull request #448 for this. I also have a step-by-step guide for this prepared to publish on my blog. Don't know if it's suitable to add it to cc26xx-web-demo README.md.

g-oikonomou, 2018-04-23:
Perfect thanks. Provide a link to your blog and we can decide on the best way to handle it.

chenek, 2018-04-23:
@g-oikonomou you can check https://sunmaysky.blogspot.tw/2018/04/how-to-connect-contiki-ng-cc26xx-web.html

g-oikonomou, 2018-04-23:
Thanks for the link. I am closing the issue now.

closed 2018-04-23 by g-oikonomou

BlueberryTech01, 2018-05-03:
Hi @g-oikonomou , I'd like to ask you something. I want to drive the LEDs of my device through cloud. Could you give me some advice?

chenek, 2018-05-03:
@BlueberryTech01 your question seems out of scope of this issue. I would suggest you to create another thread for your question.

BlueberryTech01, 2018-05-03:
Why? I only asked if with Contiki it was possible to turn on/off led with a cloud.

BlueberryTech01, 2018-05-03:
@chenek I thought I'd run this my problem with node-red. What do you think?

chenek, 2018-05-03:
I don't have experience using node-red to do this but I had connected cc26xx-web-demo to a local mosquitto mqtt server and use moqsuitto_pub to toggle LED on LAUNCHXL-CC1310 running cc26xx-web-demo.

g-oikonomou, 2018-05-03:
It should be possible to do exactly the same through IBM Watson, but only for registered devices (functionality not available on quickstart). For information as to how, I suggest reading the Watson docs, but the same publish message should do the trick.

BlueberryTech01, 2018-05-03:
@g-oikonomou I read all of Watson IoT's guidance. I thought there was something to create buttons to turn LEDs on and off from the Dashoboard

g-oikonomou, 2018-05-03:
Sorry, I cannot offer support for Watson. I have never used the registered device feature myself.

BlueberryTech01, 2018-05-04:
@chenek @g-oikonomou I talked to an IBM contributor. Then, you cannot create buttons to alter an LED or whatever. The only way to do this is to use node-red with Mosquitto and connect it to the IoT platform.

BlueberryTech01, 2018-05-25:
Node-Red works with the example CC26XX-Web-Demo