contiki-ng/contiki-ng #442: cc26xx-web-demo/cetic-6lbr-client does not support RPL Classic

simonduq, 2018-04-18:
Does simply not compile, see https://github.com/contiki-ng/contiki-ng/issues/431#issuecomment-382214734

It is a problem because 6lbr still requires RPL Classic.

g-oikonomou, 2018-04-18:
We need to fix this. What this code tries to do is to send to 6lbr the device's own preferred parent so 6lbr can display it in its web page. This was written for and used to work with classic, not sure at which point it broke.

Given how the example does some 6lbr-specific stuff, we should perhaps also test this example with classic on travis.

simonduq, 2018-04-18:
I guess this is how it broke:
* RPL Lite was added and made default
* This example was not updated to explicitly use Classic
* Then code evolved and Classic support broke

g-oikonomou, 2018-04-18:
Just to clarify, this example does _not_ need classic, it will work just fine with lite too.

For lite obviously there is no point in including the 6lbr-client, but this can be disabled using a macro so perhaps we should auto-disable when building for lite until a point when 6lbr supports lite too.

simonduq, 2018-04-18:
Oh I see. Let's do that, and then add a build test for Classic.

PR #443 (2018-04-18): Fixing cc26xx 6lbr client example
The cc26xx 6lbr client example was meant for RPL Classic.
This PR:
* Fixes compilation of the example for RPL Classic
* Only compiles the example when RPL Classic is enabled
* Adds a compile test for this example with RPL Classic

Fixes #442
closed 2018-04-22 by simonduq