contiki-ng/contiki-ng #459: A pass on all tutorials prior to 4.1

simonduq, 2018-04-26:
Basics:
* [ ] Hello, World! 
* [ ] Logging
* [ ] NG shell
* [ ] RAM and ROM usage
* [ ] Energy monitoring
* [ ] Timers and events

Networking:
* [ ] IPv6 ping
* [ ] RPL basics
* [ ] RPL with border router
* [ ] TSCH and 6TiSCH
* [ ] CoAP
* [ ] LWM2M

Simulation:
* [ ] Cooja: getting started
* [ ] Cooja: simulating a RPL network
* [ ] Cooja: simulating a RPL network with a border router
* [ ] Running Contiki-NG in Renode

simonduq, 2018-04-26:
I propose the removal of *Introduction to Contiki-NG Programming* , whose content is redundant with *Processes and events*

g-oikonomou, 2018-04-27:
Renode tutorial on Mac OS:

```
$ make renode TARGET=cc2538dk SCRIPT=rpl-udp.resc 
  CC        ../../arch/cpu/cc2538/./ieee-addr.c
  CC        ../../arch/cpu/cc2538/cc2538.lds
  CC        ../../arch/cpu/cc2538/./startup-gcc.c
  CC        udp-client.c
  LD        udp-client.elf
arm-none-eabi-objcopy -O ihex udp-client.elf udp-client.i16hex
srec_cat udp-client.i16hex -intel -o udp-client.hex -intel
arm-none-eabi-objcopy -O binary --gap-fill 0xff udp-client.elf udp-client.bin
cp udp-client.elf udp-client.cc2538dk
  CC        udp-server.c
  LD        udp-server.elf
arm-none-eabi-objcopy -O ihex udp-server.elf udp-server.i16hex
srec_cat udp-server.i16hex -intel -o udp-server.hex -intel
arm-none-eabi-objcopy -O binary --gap-fill 0xff udp-server.elf udp-server.bin
cp udp-server.elf udp-server.cc2538dk
renode rpl-udp.resc
make: renode: No such file or directory
make: *** [renode] Error 1
rm udp-client.o udp-server.o udp-client.i16hex obj_cc2538dk/startup-gcc.o udp-server.i16hex
```

PiotrZierhoffer, 2018-04-27:
This was already reported to us by Joakim earlier today. On macOS we need to manually create a symlink or an alias, as DMGs are just mounted and can't influence the rest of the filesystem.

We updated our installation instructions accordingly, and Joakim slightly updated the tutorial.

Please accept it as a temporary inconvenience. We decided this option is better than to fiddle with the relevant Makefile in contiki-ng.

g-oikonomou, 2018-04-27:
@PiotrZierhoffer Thanks Piotr.

Here is what I've done in a nutshell.

* Installed mono
* Copied renode from the `.dmg` to `Applications`.
* Created alias as per [renode/renode/README.rst](https://github.com/renode/renode/blob/master/README.rst)
```
alias renode="mono /Applications/Renode.app/Contents/MacOS/bin/Renode.exe"
```

So far so good, no problems.

Then I followed the tutorial here and got the error above.

Subsequently:

```
$ renode
2018-04-27 18:57:35.696 mono[91580:7414688] *** WARNING: Method userSpaceScaleFactor in class NSView is deprecated on 10.7 and later. It should not be used in new applications. Use convertRectToBacking: instead. 
Fatal error:
Toolkit could not be loaded
  at Xwt.Toolkit.LoadBackend (System.String type, Boolean isGuest, Boolean throwIfFails) [0x00000] in <filename unknown>:0 
  at Xwt.Toolkit.Load (System.String fullTypeName, Boolean isGuest) [0x00000] in <filename unknown>:0 
  at Xwt.Application.Initialize (System.String backendType) [0x00000] in <filename unknown>:0 
  at Xwt.Application.Initialize (ToolkitType type) [0x00000] in <filename unknown>:0 
  at Antmicro.Renode.UI.XwtProvider.Initialize () [0x00000] in <filename unknown>:0 
  at Antmicro.Renode.UI.XwtProvider+<>c__DisplayClass9_0.<StartXwtThreadOnMainThread>b__0 () [0x00000] in <filename unknown>:0 
  at Antmicro.Renode.Emulator.ExecuteAsMainThread () [0x00000] in <filename unknown>:0 
  at Antmicro.Renode.Program.Main (System.String[] args) [0x00000] in <filename unknown>:0 
Inner exception:
Exception has been thrown by the target of an invocation.
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) [0x00000] in <filename unknown>:0 
  at System.Activator.CreateInstance (System.Type type, Boolean nonPublic) [0x00000] in <filename unknown>:0 
  at System.Activator.CreateInstance (System.Type type) [0x00000] in <filename unknown>:0 
  at Xwt.Backends.ToolkitEngineBackend.CreateBackend (System.Type backendType) [0x00000] in <filename unknown>:0 
  at Xwt.Backends.ToolkitEngineBackend.CreateBackend[TextLayoutBackendHandler] () [0x00000] in <filename unknown>:0 
  at Xwt.Toolkit.Initialize (Boolean isGuest) [0x00000] in <filename unknown>:0 
  at Xwt.Toolkit.LoadBackend (System.String type, Boolean isGuest, Boolean throwIfFails) [0x00000] in <filename unknown>:0 
Inner exception:
An exception was thrown by the type initializer for Xwt.GtkBackend.GtkTextLayoutBackendHandler
  at (wrapper managed-to-native) System.Reflection.MonoCMethod:InternalInvoke (System.Reflection.MonoCMethod,object,object[],System.Exception&)
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) [0x00000] in <filename unknown>:0 
Inner exception:
libcairo.2.dylib
  at (wrapper managed-to-native) Cairo.NativeMethods:cairo_image_surface_create (Cairo.Format,int,int)
  at Cairo.ImageSurface..ctor (Format format, Int32 width, Int32 height) [0x00000] in <filename unknown>:0 
  at Xwt.GtkBackend.GtkTextLayoutBackendHandler..cctor () [0x00000] in <filename unknown>:0 
```

mono problem maybe?

PiotrZierhoffer, 2018-04-27:
This is quite surprising, as we tested it today on a relatively clean macOS.

It looks like either libcairo.2.dylib is missing or it's missing references (somehow it cannot be loaded).

Can you give me the output of the following commands?

- mono --version (mine, for example, is 5.8.1.0)
- ls /usr/local/lib/libcairo.2.dylib; ls /usr/lib/libcairo.2.dylib (in my case it is in /usr/local/lib)
- otool -L /path/to/libcairo.2.dylib (I'm not a mac expert, but if I understand it should return a list of libraries that libcairo depends on - informing of an error if some of them are not found)

Also, what is your OS version?

Unfortunately, I don't have a mac at home, only at the office - and with SSH only I cannot really bootstrap anything fresh to verify.

g-oikonomou, 2018-04-27:
macOS High Sierra fully patched with recent macOS updates

Version 10.13.4 (17E199)

```
$ mono --version
Mono JIT compiler version 4.0.4 ((detached/5ab4c0d Tue Aug 25 08:35:07 EDT 2015)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:           normal
	SIGSEGV:       altstack
	Notification:  kqueue
	Architecture:  x86
	Disabled:      none
	Misc:          softdebug 
	LLVM:          yes(3.6.0svn-mono-(detached/a173357)
	GC:            sgen
```

```
$ ls /usr/local/lib/libcairo.2.dylib
/usr/local/lib/libcairo.2.dylib
```

```
$ ls /usr/lib/libcairo.2.dylib
ls: /usr/lib/libcairo.2.dylib: No such file or directory
```

```
$ otool -L /usr/local/lib/libcairo.2.dylib
/usr/local/lib/libcairo.2.dylib:
	/usr/local/lib/libcairo.2.dylib (compatibility version 11203.0.0, current version 11203.16.0)
	/usr/local/lib/libpixman-1.0.dylib (compatibility version 33.0.0, current version 33.6.0)
	/usr/local/lib/libfontconfig.1.dylib (compatibility version 10.0.0, current version 10.0.0)
	/usr/local/lib/libfreetype.6.dylib (compatibility version 18.0.0, current version 18.2.0)
	/usr/local/lib/libpng16.16.dylib (compatibility version 30.0.0, current version 30.0.0)
	/opt/X11/lib/libxcb-shm.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/opt/X11/lib/libxcb-render.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/opt/X11/lib/libxcb.1.dylib (compatibility version 3.0.0, current version 3.0.0)
	/opt/X11/lib/libXrender.1.dylib (compatibility version 5.0.0, current version 5.0.0)
	/opt/X11/lib/libX11.6.dylib (compatibility version 10.0.0, current version 10.0.0)
	/opt/X11/lib/libXext.6.dylib (compatibility version 11.0.0, current version 11.0.0)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.5)
	/System/Library/Frameworks/ApplicationServices.framework/Versions/A/ApplicationServices (compatibility version 1.0.0, current version 48.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1213.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1151.16.0)
	/System/Library/Frameworks/CoreGraphics.framework/Versions/A/CoreGraphics (compatibility version 64.0.0, current version 600.0.0)
```

I'm about try with a more recent mono to prevent wasting your time. I think I installed mono when there was a project called "emul8" around...

simonduq, 2018-04-27:
Hi @PiotrZierhoffer 

Am also trying out the tutorial now, and facing a different issue.  The client just prints
```
[INFO: App       ] Not reachable yet
```

once and then nothing (it should at least print it every minute).

On the other hand, the terminal from where Renode is running suggests the simulation goes on normally, at least there is traffic.

Also:
1. I tried enabling the shell. Is serial input supported?
2. In the tutorial, I didn't get *Starting a single node requires the following command: machine Start.*

Thanks!

g-oikonomou, 2018-04-27:
@PiotrZierhoffer thanks

I have updated mono:

```
$ mono --version
Mono JIT compiler version 5.10.1.47 (2017-12/8eb8f7d5e74 Fri Apr 13 20:18:12 EDT 2018)
[...]
```

I ran `renode` and in the console:
```
i @<path>/contiki-ng/examples/rpl-udp/rpl-udp.resc
start
pause
```

All fine. However:

```
$ make renode TARGET=cc2538dk SCRIPT=rpl-udp.resc
[...]
renode rpl-udp.resc
make: renode: No such file or directory
make: *** [renode] Error 1
rm udp-client.o udp-server.o udp-client.i16hex obj_cc2538dk/startup-gcc.o udp-server.i16hex
```

Looks like it perhaps doesn't like the alias. Shameless ping to @joakimeriksson too

joakimeriksson, 2018-04-27:
True - I just tried with the make and it seems not to like the alias but will run if I replace with mono <PATH-TO-Renode> in the makefile. So alias seems not so good.

nfi, 2018-04-27:
`make` can not use bash aliases in any simple way.

1. Instead of the alias, we could document how to create a small bash script named `renode` that starts renode. This script would need to be installed in the user command search path.

2. We could change the Makefile to detect if running under OSX and let it start renode from the standard installation path `/Applications/Renode.app/Contents/MacOS/bin/Renode.exe`. If `Renode.exe` is not found, `make` can ask the user to correct the path. And if not running under OSX, `make` can simply call the `renode` command as before.

g-oikonomou, 2018-04-28:
@nfi could [something line this](https://github.com/g-oikonomou/contiki-ng/commit/bd109ad3b66d1a8fc5b5276f38fe45fad5ecbd55) be what we want?

yatch, 2018-04-28:
@g-oikonomou I ran through [Toolchain installation on OS X](https://github.com/contiki-ng/contiki-ng/wiki/Toolchain-installation-on-OS-X) and found `tuntap` cannot be installed via `brew`.

```bash
$ brew install tuntap
Error: No available formula with the name "tuntap"
It was migrated from homebrew/core to caskroom/cask.
You can access it again by running:
  brew tap caskroom/cask

$ brew tap caskroom/cask

$ brew install tuntap
Error: No available formula with the name "tuntap"
It was migrated from homebrew/core to caskroom/cask.
You can access it again by running:
  brew tap caskroom/cask
```

http://brewformulas.org/tuntap shows this package is "inactive"... What should we do?

yatch, 2018-04-28:
@g-oikonomou !!! sorry for the stupid question... It works!
```bash
$ brew cask install tuntap
...
🍺  tuntap was successfully installed!
```
Thank you. I updated the wiki page with this `brew cask install tuntap`.

PiotrZierhoffer, 2018-04-28:
As @nfi suggests, we hurried with a suboptimal solution. We will fix it soon.

@simonduq - unfortunately I cannot tell you anything at the moment, as during the weekend I won't have access to my PC. I will try to verify if as soon as possible.

simonduq, 2018-04-28:
Thanks! Some details below:
```bash
$ uname -a
Linux ubuntu 4.13.0-39-generic #44~16.04.1-Ubuntu SMP Thu Apr 5 16:43:10 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

```bash
$ mono --version
Mono JIT compiler version 5.10.1.47 (tarball Tue Apr 17 15:12:35 UTC 2018)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
        TLS:           __thread
        SIGSEGV:       altstack
        Notifications: epoll
        Architecture:  amd64
        Disabled:      none
        Misc:          softdebug 
        Interpreter:   yes
        LLVM:          supported, not enabled.
        GC:            sgen (concurrent by default)
```

And a screenshot, taken several (real-time, not sure about emulated time) minutes in:

[image]

nfi, 2018-04-28:
@g-oikonomou The makefile with renode variable should work nicely if we go for alt. 2. Also nice that the `cc2538dk` specifics was moved to its own file but perhaps the `SCRIPT` variable still should include the file postfix in case it is specified as a make argument (`make SCRIPT=rpl-udp.resc renode`). What do you think?

@simonduq I am also using mono 5.10.1.47 but under OSX where it seems to work for me so I am not sure what is wrong with the node output when you run the example. I will try to run the test under Linux also and see if I get the same result.

```
[INFO: CC2538DK  ] TI SmartRF06 + cc2538EM
[INFO: App       ] Not reachable yet
[INFO: App       ] Sending request 0 to fd00::200:0:0:1
[INFO: App       ] Received response 0 (tc:0) from fd00::200:0:0:1
```

g-oikonomou, 2018-04-28:
@nfi 

> but perhaps the SCRIPT variable still should include the file postfix

Yes, good point. I am also wondering whether this entire logic can / should be moved to the CC2538DK Makefile, in prep of getting more examples supported.

nfi, 2018-04-28:
I think it would be very nice to move the logic to the CC2538DK Makefile. Then we would only need to add a script file to add renode support for an example without any need of Makefile modifications.

simonduq, 2018-04-28:
@nfi is there a way I can look at pcap to check if traffic looks fine? Can I also check elapsed simulation time?

simonduq, 2018-04-28:
Thanks @PiotrZierhoffer , let me try this.

For `machine *` commands, I must be missing something. Where am I supposed to type these? Not the main terminal ('monitor') it seems.

PiotrZierhoffer, 2018-04-28:
Yes, you have to be in a machine context. Type in `mach`, this will give you a list of machines with their IDs. `mach set 0` to set the context of the first one.

PiotrZierhoffer, 2018-04-29:
@simonduq I do believe you might be observing a performance issue. The CC nodes are not sleepy, and Contiki has some strict timeout requirements - so the nodes are synchronized quite frequently. It causes the nodes to run at 25% of their target performance on a virtual machine on my laptop (not a very strong machine).

I just ran the same app with reduced `START_INTERVAL` and `SEND_INTERVAL` (to 5s each) and I observed a correct communication much faster. With the unmodified code it takes about 4mins. You can verify the real elapsed time with `machine ElapsedHostTime` (just like before - after `mach set 0`).

simonduq, 2018-04-29:
Did more work on the Renode tutorial: https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-Running-Contiki%E2%80%90NG-in-Renode

Might need adjustments for OSX? (e.g. regarding the wireshark addgroup comment)