contiki-ng/contiki-ng #470: NAT64 and DNS64 tools

simonduq, 2018-04-27:
We need better tools to use NAT64 and DNS64, for example:
* DNS64 client: last time I tried, did not work
* DNS64 shell command: one should be able to do a lookup from command line
* DNS64 in existing shell commands: one should be able to target hosts from their name
* NAT64 in existing shell commands: one should be able to target hosts with their IPv4 address or IPv4-compatible IPv6 (e.g. ::ffff:192.0.2.128)

joakimeriksson, 2018-04-27:
👍 agree 100%

simonduq, 2018-10-19:
This is not going to happen in the near future, flagging as long-term

joakimeriksson, 2019-10-04:
I will add a easy to use NAT64/DNS64 example soon. With support in NBR and Native platform.

fkerem, 2021-04-28:
Hi,

What is the status on NAT64 support and Jool version 4 ?
Any updates?

Thanks.

nvt, 2026-09-30:
NAT64 support has been improved considerably recently through the nat64 and ip64 modules. Closing.

closed 2026-09-30 by nvt