contiki-ng/contiki-ng #472: Cannot define make rules in MODULE makefiles

g-oikonomou, 2018-04-28:
In the top-level `Makefile.include` we do this:

```
include $(MODULES_INCLUDES)
# Iterate once more: include the modules added from the previous include.
# Only works with one level of nested module inclusion.
include $(MODULES_INCLUDES)
```

The second invocation of `include` will go and re-include the makefiles that were present in `MODULES_INCLUDES` at the time of first invocation.

This is problematic: Having a rule in one of these makefiles will result in a Make warning message:

```
../../os/services/rpl-border-router/Makefile.rpl-border-router:11: warning: overriding commands for target `test'
../../os/services/rpl-border-router/Makefile.rpl-border-router:11: warning: ignoring old commands for target `test'
```

We need a better way to allow a module to include another module.

simonduq, 2018-10-19:
A simple approach is to let modules define rules in a distinct makefile. We already have `Makefile.[module]`, we could add an optional `Makefile-rules.[module]` and then do:
```diff
diff --git a/Makefile.include b/Makefile.include
index 3578730..ea3328d 100644
--- a/Makefile.include
+++ b/Makefile.include
@@ -227,6 +227,8 @@ include $(MODULES_INCLUDES)
 # Iterate once more: include the modules added from the previous include.
 # Only works with one level of nested module inclusion.
 include $(MODULES_INCLUDES)
+MODULES_RULES_INCLUDES = ${wildcard ${foreach d, $(MODULEDIRS), $(d)/Makefile-rules.${notdir $(d)}}}
+include $(MODULES_RULES_INCLUDES)
 
 # C-include module-specific macros using -imacros
 MODULES_IMACROS = ${wildcard ${foreach d, $(MODULEDIRS), $(d)/module-macros.h}}
```

simonduq, 2018-10-19:
Labeling as long-term, no reason to rush this one