contiki-ng/contiki-ng #499: RPL nodes periodically disconnected when using TSCH

ianno, 2018-05-09:
I tried running the simulation in `examples/rpl-udp/rpl-udp-cooja.csc` using nodes with TSCH enabled.
In this scenario, nodes will periodically lose their connection to the network (recurrent 'Not reachable yet' messages).
I'm pasting the modified Makefile (`examples/rpl-udp/Makefile`):

```
CONTIKI_PROJECT` = udp-client udp-server
all: $(CONTIKI_PROJECT)

CONTIKI=../..

MAKE_MAC = MAKE_MAC_TSCH
include $(CONTIKI)/Makefile.include
```

ianno, 2018-05-09:
It seems the issue is with the RPL LITE driver.
Using `MAKE_ROUTING = MAKE_ROUTING_RPL_CLASSIC` in the Makefile makes the simulation working without problems.

simonduq, 2018-05-09:
Hey I believe the issue to be specific to Cooja motes, where there are known sync issues. But I see your added comment, definitely needs investigating more.

ianno, 2018-05-09:
Thanks for the super fast reply.
If it can help, this is what happens:

```
07:49.735	[WARN: TSCH      ] ! EB JP too high 36, leaving the network	ID:13
07:49.735	[INFO: TSCH-LOG  ] {asn 00.0000b757 link  0   7   0  0 ch 20} drift 0	ID:13
07:49.735	[INFO: TSCH-LOG  ] {asn 00.0000b757 link  0   7   0  0 ch 20} bc-0-0 rx LL-000c->LL-NULL, len  35, seq  60, edr   0, dr   0	ID:13
07:49.735	[WARN: TSCH      ] leaving the network, stats: tx 16, rx 13, sync 1	ID:13
07:49.735	[INFO: TSCH Queue] update time source: 000c.000c.000c.000c -> (NULL LL addr)	ID:13
07:49.735	[WARN: RPL       ] local repair (TSCH leaving)	ID:13
07:49.735	[INFO: RPL       ] reset MRHOF	ID:13
07:49.735	[INFO: RPL       ] removing all neighbors	ID:13
07:49.735	[INFO: RPL       ] parent switch: fe80::20c:c:c:c -> (NULL IP addr)	ID:13
07:49.735	[WARN: RPL       ] no parent, scheduling periodic DIS, will leave if no parent is found	ID:13
07:49.735	[INFO: RPL       ] nbr: own state, addr fd00::20d:d:d:d, DAG state: initialized, MOP 1 OCP 1 rank 65535 max-rank 6262, dioint 15, nbr count 0 (Parent switch)	ID:13
07:49.735	[INFO: RPL       ] nbr: end of list	ID:13
07:50.306	[WARN: RPL       ] rpl_icmp6_dao_output: no preferred parent, skip sending DAO	ID:13
07:51.737	[INFO: TSCH      ] scanning on channel 15	ID:13
```

simonduq, 2018-05-09:
OK, am looking into this. It actually is not related to TSCH but rather RPL Lite in congested environment. Here we're congested because all nodes in Cooja are in interference range (each Tx interferes all 15 neighbors), and because of TSCH, in the default config with a slotframe of 7, gives each node only 14 chances to send each second. Setting a slotframe length of 3 instead, or reducing the interference range, solves the issue immediately (network running without any App loss nor disconnection after 40 minutes simulated time).

The issue also exists in RPL Classic, to a lesser extent. The reason it appears mostly on Lite is because there are mechanisms meant to recover from faulty situations, through added control traffic. Here we have an escalation of such traffic, increasing congestion further, in a vicious circle. I'm sure things can be improved, we need to make the mechanisms more conservative, include rate-limitation to them. What I have identified so far:
* Urgent probing happens too fast, so fast that several probes end up in queue at the same time (pb with Lite and Classic)
* DIS broadcast are sent every 30s as long as we haven't settled on a parent, which we don't get a chance to do given the poor network conditions (only Lite)

Looks like this is not all, will need more investigation. First need to fix the two above. Thanks a lot for reporting!

ianno, 2018-05-09:
Of course. Happy to help!

simonduq, 2018-05-09:
OK, one more mechanism that seems to play very poorly under such congestion is squared ETX in MRHOF, disabled by default in Classic but enabled in Lite. Thinking we actually want it disabled for both, by default.

simonduq, 2018-05-09:
So, with the two PRs above you should be good.
https://github.com/contiki-ng/contiki-ng/pull/503 is even probably sufficient to get things working.

ianno, 2018-05-09:
That was fast:-) Awesome.
Thanks a lot!

simonduq, 2018-05-12:
Both PRs now merged, closing.

closed 2018-05-12 by simonduq