contiki-ng/contiki-ng #514: ./../os/net/ipv6/uip-ds6.c:94:22: error: ‘iid_prefix’ defined but not used

cuongdodinh, 2018-05-13:
Dear All,

I run in Raspi, How can I fix this error?

make TARGET=native all
  CC        ../../os/net/ipv6/uip-ds6.c
../../os/net/ipv6/uip-ds6.c:94:22: error: ‘iid_prefix’ defined but not used [-Werror=unused-const-variable=]
 static const uint8_t iid_prefix[] = { 0x00, 0x00 , 0x00 , 0xff , 0xfe , 0x00 };
                      ^~~~~~~~~~
cc1: all warnings being treated as errors
../../Makefile.include:317: recipe for target 'obj_native/uip-ds6.o' failed
make: *** [obj_native/uip-ds6.o] Error 1

@chenek  Would you like help me?

Thanks & B.R,

chenek, 2018-05-13:
@cuongdodinh I suppose you can remove claim iid_prefix in uip-ds6.c to fix the issue.

simonduq, 2018-05-13:
What concerns me here is that this wasn't caught by the CI suite. Which compiler are you on? It looks like almost all compiled (all in our test suite) do not issue unused variable warnings on `const`. Ideally we should first reproduce in Docker, add a test and then fix.

cuongdodinh, 2018-05-13:
Dear Simonduq,
I use Raspi 3. Downloaded Contili-NG on it. Then compiler in Raspi 3.

Thank you.

simonduq, 2018-05-13:
What's your compiler version?

cuongdodinh, 2018-05-14:
Dear SemonDuq,

Raspbian 6.3.0.0 -18+raspi1 + deb9u1 (20170516)

simonduq, 2018-05-14:
Please provide the full output, e.g.
```
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.6/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Debian 4.6.3-14+rpi1' --with-bugurl=file:///usr/share/doc/gcc-4.6/README.Bugs --enable-languages=c,c++,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.6 --enable-shared --enable-linker-build-id --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.6 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --enable-plugin --enable-objc-gc --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.6.3 (Debian 4.6.3-14+rpi1) 
```

cuongdodinh, 2018-05-14:
Dear SimonDuq,
Here you are.
pi@raspberrypi:~ $ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/6/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 6.3.0-18+rpi1+deb9u1' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1+deb9u1)

cuongdodinh, 2018-05-15:
Dear Simonduq,
Thank you very much.

closed 2018-05-15 by cuongdodinh