contiki-ng/contiki-ng #520: cc26xx-web-demo not joins rpl-border-router on Raspi

cuongdodinh, 2018-05-15:
Dear All,
I build cc26xx-web-demo running on LAUNCHXL-CC1310 
rpl-border-router build on Raspberry Pi with slip radio running on LAUNCHXL-CC1310
as below. I followed this blog from @chenek 
https://sunmaysky.blogspot.com/2018/04/how-to-build-and-run-contiki-ng-cc26xx.html

How can I fix it? Green Led blink fast

sudo ./border-router.native fd00::1/64 -s /dev/ttyACM0
[INFO: Main      ] Starting Contiki-NG-release/v4.1-132-g07ddf5026-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: br-mac
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 25
[INFO: Main      ] Link-layer address 0102.0304.0506.0708
[INFO: Main      ] Tentative link-local IPv6 address fe80::302:304:506:708
[INFO: Native    ] Added global IPv6 address fd00::302:304:506:708
[INFO: RPL BR    ] Contiki-NG Border Router started
[INFO: BR        ] RPL-Border router started
********SLIP started on ``/dev/ttyACM0''
opened tun device ``/dev/tun0''
ifconfig tun0 inet `hostname` up
ifconfig tun0 add fd00::1/64
ifconfig tun0

tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 127.0.1.1  netmask 255.255.255.255  destination 127.0.1.1
        inet6 fe80::ac3c:7856:63a2:9ada  prefixlen 64  scopeid 0x20<link>
        inet6 fd00::1  prefixlen 64  scopeid 0x0<global>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

[INFO: BR        ] Setting prefix fd00::1
[INFO: BR        ] Server IPv6 addresses:
[INFO: BR        ]   fd00::212:4b00:a27:cfc3
[INFO: BR        ]   fe80::212:4b00:a27:cfc3

#0012.4b00.0a27.cfc3> ip-nbr
Node IPv6 neighbors: none
#0012.4b00.0a27.cfc3> 

@chenek  Would you like helpme?

chenek, 2018-05-15:
It looks your rpl-boarder-router runs. What is the problem?

cuongdodinh, 2018-05-15:
Dear @chenek ,
cc26xx-web-demo running on LAUNCHXL-CC1310 not join into RPL.

#12.4b00.0a27.cfc3> ip-nbr
Node IPv6 neighbors: none
#12.4b00.0a27.cfc3>

chenek, 2018-05-15:
Make sure you have cc26xx-web-demo and rpl-border-Router Running on the same channel.

cuongdodinh, 2018-05-16:
Dear @chenek ,

In slip-radio folder, I added #define RF_CORE_CONF_CHANNEL 25 in project-conf.h,
But when print to screen, it's still show chanel 0;

[INFO: Main      ] Starting Contiki-NG-release/v4.1-135-g7f975db-dirty
                                                                      [INFO: Main      ] - Routing: nullrouting
                               [INFO: Main      ] - Net: slipnet
                                                                [INFO: Main      ] - MAC: CSMA
              [INFO: Main      ] - 802.15.4 PANID: 0xabcd
                                                         [INFO: Main      ] - 802.15.4 Default channel: 0
                         [INFO: Main      ] Link-layer address 0012.4b00.0a27.cfc3
  [INFO: Main      ] Tentative link-local IPv6 address fe80::212:4b00:a27:cfc3
                                                                              _**[INFO: CC26xx/CC13xx] TI CC1310 LaunchPad
                                       [INFO: CC26xx/CC13xx]  RF: Channel 0**_
                                                                           [INFO: CC26xx/CC13xx]  Node ID: 53187
Would you like help me.

Thanks&B.R

chenek, 2018-05-17:
Try to use "define IEEE802154_CONF_DEFAULT_CHANNEL      25" instead of "#define RF_CORE_CONF_CHANNEL 25" in project-conf.h

cuongdodinh, 2018-05-18:
Dear @chenek,

Thank for your help. I change define in arch #define IEEE802154_CONF_DEFAULT_CHANNEL                      25
Now it is working.
Thank you.

closed 2018-05-18 by cuongdodinh

chenek, 2018-05-18:
I also revise the content in https://sunmaysky.blogspot.com/2018/04/how-to-build-and-run-contiki-ng-cc26xx.html in case someone will have the same issue again when he/she follows the steps.

cuongdodinh, 2018-05-18:
Dear @chenek ,
Thank you.