contiki-ng/contiki-ng #521: Docker on OSX requires different setup than linux

dcsouthwick, 2018-05-15:
OSX does not address USB devices via /dev/bus/usb but though the hypervisor. Attempting to use the linux Docker instructions will fail with somewhat misleading
```The path /dev/bus/usb
is not shared from OS X and is not known to Docker.
You can configure shared paths from Docker -> Preferences... -> File Sharing.```
that the user might share /dev/bus/usb via docker preferences (in fact, it does not exist on OSX). 

I've written a wiki stub that might help others who are using macbooks:
https://github.com/contiki-ng/wiki-sandbox/wiki/Docker-on-OSX

Please modify/edit/change if I've got something wrong here or out of place.

yatch, 2018-05-15:
Nice. Yes, it'd be better to have "on macOS" section in [Docker](https://github.com/contiki-ng/contiki-ng/wiki/Docker) wiki page at least.

A couple of questions:

* Your instruction is for "Docker Toolbox on macOS", isn't it...? I'm using Docker For Mac, that doesn't support USB pass-through...
* Why do you need socat? "Allow connections from network clients" option in "Security" tab of "X11 Preferences" doesn't work in this case?

dcsouthwick, 2018-05-15:
Hi - yes, at closer inspection I am using docker-machine, which is the "old" docker provided from Docker Toolbox. Regarding socat, I'm not sure but simple x11 didnt work for me. I found it from https://github.com/moby/moby/issues/8710

yatch, 2018-05-16:
@dcsouthwick Thanks! Now I'm looking into X11 with Docker on macOS. I'll get back to you later.

yatch, 2018-07-24:
@dcsouthwick I've updated the wiki page: https://github.com/contiki-ng/contiki-ng/wiki/Docker

Thank you for your suggestions!

closed 2018-07-24 by yatch