contiki-ng/contiki-ng #526: cc26xx-web-demo MQTT disconnected when I use mosquitto_pub to publish topic to IBM Watson IoT

chenek, 2018-05-17:
I setup IBM Watson IoT and make my LAUNCHXL-CC1310 running cc26xx-web-demo. There is no problem to see cc26xx-web-demo publish MQTT messages, such as device status, to IBM Watson IoT. However, when I try to use the following mosquitto_pub command to do publish to control LED on LAUNCHXL-CC1310, I see "MQTT - Disconnected by tcp event 1"

`mosquitto_pub -h 169.45.2.20 -u use-token-auth -P <token> -i d:uc6bmi:cc26xx-web-demo:00124b27f319 -m "1" -t iot-2/cmd/leds/fmt/json"`

The followings are my UART log of cc26xx-web-demo when subscribing to IBM Watson Iot

MQTT - Finished reading packet!
MQTT - total data was 4 bytes of data.
MQTT - Got SUBACK
APP - Application is subscribed to topic successfully
MQTT - Done in send_subscribe!
MQTT - Disconnected by tcp event 1
APP - MQTT Disconnect. Reason 1
MQTT - Abort
Disconnected
Disconnected. Attempt 3 in 2048 ticks

I had done the same test using moatuitto server running on my 6lbr and use the same mosquitto_pub command to do publish to control LED on LAUNCHXL-CC1310. It works fine.
`
mosquitto_pub -h bbbb::e786:9d85:9446:709 -m "1" -t iot-2/cmd/leds/fmt/json`

The followings are my UART log of cc26xx-web-demo when subscribing to local mosquitto MQTT server

MQTT - Got TCP_DATA_SENT
MQTT - Got TCP_DATA_SENT
tcp_input with 28 bytes of data:
MQTT - Read VHDR '30'
MQTT - Read Remaining Length byte
MQTT - Finished reading remaining length byte
MQTT - Read PUBLISH topic len 23
MQTT - topic_pos: 0 copy_bytes: 23MQTT - Got topic 'iot-2/cmd/leds/fmt/json'- Copied 1 payload bytes
MQTT - Copied bytes:
30

MQTT - Finished reading packet!
MQTT - total data was 27 bytes of data.
MQTT - Got PUBLISH, called once per manageable chunk of message.
MQTT - Handling publish on topic 'iot-2/cmd/leds/fmt/json'
MQTT - This chunk is 1 bytes
APP - Application received a publish on topic 'iot-2/cmd/leds/fmt/json'. Payload size is 1 bytes. Content:

Pub Handler: topic='iot-2/cmd/leds/fmt/json' (len=23), chunk_len=1
MQTT - (handle_publish) resetting packet.
MQTT - Call to mqtt_publish...
MQTT - Accepted!
APP - Publish!
Publishing
MQTT - Got mqtt_do_publish_mqtt_event!
MQTT - Sending publish message! topic iot-2/evt/status/fmt/json topic_length 25

As you can see, cc26xx-web-demo can get "tcp_input with 28 bytes of data:" when I do publish to local mosquitto MQTT server. If I do publish to IBM Watson Iot, it will get "MQTT - Disconnected by tcp event 1" and "APP - MQTT Disconnect. Reason 1".

After surfing some IBM documents, I suspect they don't support device command in format such as iot-2/cmd/leds/fmt/json. Does anyone know if this is the root cause and how to fix this?

g-oikonomou, 2018-05-17:
https://console.bluemix.net/docs/services/IoT/devices/mqtt.html#mqtt

> Devices cannot subscribe to events from other devices. A device receives commands that are published only to its own device.

I'm not sure what I means. I suspect your problem could be due to the client ID in the publish message.

This seems to be related to how you are using Watson.

chenek, 2018-05-18:
@g-oikonomou I don't subscribe cc26xx-web-demo running on LAUNCHXL-CC1310 to events from other devices so I don't think this is the issue. I try to use mosquitto_sub to subscribe to "iot-2/cmd/leds/fmt/json" and I also cannot get mosquitto_pub to publish to the same topic. I suspect it's format issue on topic  "iot-2/cmd/leds/fmt/json"

g-oikonomou, 2018-05-18:
So, let me try to make sure I understand your setup

* You are using IBM Watson
* You are using mosquitto to subscribe to a topic
* You are using mosquitto to publish to the same topic
* It doesn't work
* You are raising a Contiki-NG issue about this

What have I missed? :)

chenek, 2018-05-18:
Actually, it’s CC26xx-web-demo subscribes to IBM IoT platform and I use mosquitto_pub to the topic. Then, I see CC26xx-web-demo tcp disconnection— "MQTT - Disconnected by tcp event 1" and "APP - MQTT Disconnect. Reason 1".

g-oikonomou, 2018-05-18:
> I try to use mosquitto_sub to subscribe to "iot-2/cmd/leds/fmt/json" and I also cannot get mosquitto_pub to publish to the same topic. 

What did I misunderstand?

chenek, 2018-05-18:
I just do this step to verify if there’s problem in topic format.

g-oikonomou, 2018-05-18:
OK but if a mosquitto pair cannot sub/pub to each other, we don't have a bug in the code, we are obviously doing something wrong with Watson. I believe Watson has things called "devices" "applications" and whatnot. This is what the first `d:` signifies in the client identifier. I am wondering whether this is causing trouble.

There is nothing wrong with the topic.

Try using actual JSON data as the pub payload

chenek, 2018-05-18:
I try it and I see the same disconnect issue. :-(

joakimeriksson, 2026-10-21:
Closing: the broker log shows that IBM Watson rejected the publish because of its topic/authorization rules, so this is a broker-side configuration issue, not a Contiki-NG bug. If you still see problems with a standard broker on current develop, please open a GitHub Discussion.

closed 2026-10-21 by joakimeriksson