contiki-ng/contiki-ng #530: ext-flash with SensorTag cc2650

AntonioLangiu, 2018-05-17:
I'm trying to use the external flash with the SensorTag cc2650. After several tries I compiled the `cc26xxx-web-demo` example and I noticed that the new interface to interact with the SPI flash is failing also there:

```
[INFO: Main      ] Starting Contiki-NG-release/v4.1-138-g458178513-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Link-layer address 0012.4b00.0afe.bd04
[INFO: Main      ] Tentative link-local IPv6 address fe80::212:4b00:afe:bd04
[INFO: CC26xx/CC13xx] TI CC2650 SensorTag
[INFO: CC26xx/CC13xx]  RF: Channel 26, PANID 0xABCD
[INFO: CC26xx/CC13xx]  Node ID: 48388
CC26XX Web Demo Process
CC26XX Web Server
CC26XX CoAP Server
CC26XX Net UART Process
Could not open flash to load config
```

After some debugging I noticed that the failure is happening [here](https://github.com/contiki-ng/contiki-ng/blob/develop/arch/dev/ext-flash/ext-flash.c#L223).

Is there something I'm doing wrong?

g-oikonomou, 2018-05-17:
How are you getting the tag to print debug output for you?

g-oikonomou, 2018-05-17:
I'm also curious as to the values of `rbuf[0]` and `rbuf[1]`. Can you add a print for them in the fail scenario, just before the `return VERIFY_PART_POWERED_DOWN;`?

chenek, 2018-05-18:
The CC2650 on the SensorTag uses the same I/O pin for SPI SCLK signal and JTAG TDI signal. The current firmware version of the XDS110 does not support the 2-pin c-JTAG standard, so it needs to use TDI and TDO in addition to TCK and TMS (full 4-pin JTAG). Due to this shared resource, it is not possible to access the external serial flash while debugging the SensorTag with the Debug DevPack. You can refer to http://processors.wiki.ti.com/index.php/Debug_DevPack_User_Guide

AntonioLangiu, 2018-05-18:
@g-oikonomou I'm using the Debug DevPack. The values of `rbuf[0]` and `rbuf[1]` are `0xEF` and `0x12` respectively.

g-oikonomou, 2018-05-18:
OK so what @chenek is the primary suspect here.

I'd also like to double-check that the DID and MID defined in [`board.h`](https://github.com/contiki-ng/contiki-ng/blob/develop/arch/platform/srf06-cc26xx/sensortag/cc2650/board.h) are correct for your device. We did test them, but I believe there have been older sensortags using a different flash part, so could yours be one of them?

```
#define EXT_FLASH_DEVICE_ID         0x14
#define EXT_FLASH_MID               0xC2
```

AntonioLangiu, 2018-05-18:
Yep! I think this is the problem! I fixed it with this patch:
```
 --- a/arch/platform/srf06-cc26xx/sensortag/cc2650/board.h
 +++ b/arch/platform/srf06-cc26xx/sensortag/cc2650/board.h
 @@ -149,8 +149,8 @@
  #define EXT_FLASH_SPI_PIN_MISO      18
  #define EXT_FLASH_SPI_PIN_CS        14

 -#define EXT_FLASH_DEVICE_ID         0x14
 -#define EXT_FLASH_MID               0xC2
 +#define EXT_FLASH_DEVICE_ID         0x12
 +#define EXT_FLASH_MID               0xEF

  #define EXT_FLASH_PROGRAM_PAGE_SIZE 256
  #define EXT_FLASH_ERASE_SECTOR_SIZE 4096
 --
 2.13.1
```

g-oikonomou, 2018-05-18:
Ah we cross posted there... Please change board.h with those values. Looks like your device is wearing a Winbond... Just out of curiosity, how old is your tag? Does it have a revision sticker on it?

g-oikonomou, 2018-05-18:
It's a W25X40CL (4Mbit).

AntonioLangiu, 2018-05-18:
Yes, it has a revision sticker on it and says `Rev: 1.3.0` and `HW Rev 1.2.0`. (I don't know how old is it since I did not directly buy it, but I'll ask).

g-oikonomou, 2018-05-18:
It's an older one. New ones have the MX25R8035F (8MBit).

What a mess... I need to think about this.

g-oikonomou, 2018-05-18:
@AntonioLangiu please try #534 

Simply defining `SENSORTAG_CC2650_REV_1_2_0` as 1 in your `contiki-conf.h` or `project-conf.h` should do it for you.

AntonioLangiu, 2018-05-18:
I tested it with the `cc26xxx-web-demo` example and I confirm that it fixes the problem.

PR #534 (2018-05-18): Differentiate flash part across sensortag/cc2650 versions
Older CC2650 sensortags have a Winbond W25X40CL 4MBit SPI flash. Newer ones have a Macronix MX25R8035F 8MBit part. The parts have almost identical instruction sets and are both supported by our generic external flash driver. The key difference between the two parts is that they have different Manufacturer and Device IDs.

This commit allows us to switch between the two: We default to the new revision, and allow users to build for the older one by defining `SENSORTAG_CC2650_REV_1_2_0` to 1.

Current tags are version 1.5.1. Older tags are [... cut]
closed 2018-05-18 by g-oikonomou