contiki-ng/contiki-ng #536: re-mote temperature

kmartinez, 2018-05-21:
an example would be useful of using the sht25 code (in zoul/dev) as the remote-revb temperature sensor? 
maybe its a simple thing missing in the config for that platform?
Thanks!

g-oikonomou, 2018-05-21:
@kmartinez Hi Kirk great seeing you here. Is this not suitable? https://github.com/contiki-ng/contiki-ng/blob/develop/examples/platform-specific/zoul/test-sht25.c

kmartinez, 2018-05-21:
Hi George
I just wanted to add the sht to the coap demo - and have a feeling that a couple of #defines brings in the right code into the standard temperature sensor interface - via 
https://github.com/contiki-ng/contiki-ng/blob/develop/examples/coap/coap-example-server/resources/res-temperature.c

g-oikonomou, 2018-05-21:
Sure this resource exposes a CoAP resource linked to a sensor symbol called `temperature_sensor`. Not quite sure if the zoul/remote exposes the SHT with a symbol of that name. Should be pretty straightforward to do if not.

Also, see what we want to do moving forward #374

closed 2018-08-17 by g-oikonomou