contiki-ng/contiki-ng #538: GPIO API

joakimeriksson, 2018-05-23:
I am currently porting the GPIO API to the EFR32. But I discovered that the current GPIO hal is a bit hard to implement as there is potentially up to 160 GPIOs and my device happen to have 65 GPIOs.

My suggestion is to change the API slightly - to focus the API on pins and avoid the masks. If we do that we can allow any type of pin type including a struct type or introduce Port and Pin that seems very common to more or less all platforms. I would also like us to consider having slightly shorter names on the functions (if that is an option).

gpio_hal_write(pin, value)
gpio_hal_read(pin, *value)
...

Where pin is either just an int as now or for the EFR32 I would pick a struct like:
struct efr32pin {
  uint8_t port;
  uint8_t pin;
}

Then a typedef on top of that that will be used for the EFR32.

I will try to implement something for the EFR32 that feels easy to configure on the platform, and take a look at CC2538 drivers to see that this will not make things very hard on other platforms.

g-oikonomou, 2018-05-23:
I cannot see the reason why change the current pin representation from a simple number to a port.pin struct. Common perhaps, but not universal.

The problem is the masking. It applies to functions that manipulate multiple pins in one go (which we can probably live without) but also to interrupt handling logic. This latter we'll need to think about very carefully. This masking is also critical for the operating of the button HAL.

Perhaps a simple solution while you are porting is to allow those masks for some 64 of your pins and leave the rest to the platform code?

We should also think about the possibility of `typedef`ing `gpio_hal_pin_mask_t` to a struct pointer. This can be universal if we need it to be (makes `gpio_hal_event_handler` easier to managge, I think), or we can leave it to the platform but then we'll need more flexibility in the  code under `os/`

g-oikonomou, 2018-05-23:
Or those masks could simply be a `uint8_t` array and then the platform tells us how many elements...

joakimeriksson, 2018-08-16:
I have looked at the Zephyr OS API:s for drivers and specifically GPIO - it looks like:

 int gpio_pin_write(struct device *port, u32_t pin, u32_t value)
 int gpio_pin_read(struct device *port, u32_t pin, u32_t *value)
 int gpio_pin_configure(struct device *port, u32_t pin, int flags)

 int gpio_port_write(struct device *port, u32_t value)
 int gpio_port_read(struct device *port, u32_t *value)
 int gpio_port_configure(struct device *port, int flags)

 int gpio_add_callback(struct device *port, struct gpio_callback *callback)
 int gpio_pin_enable_callback(struct device *port, u32_t pin)
 int gpio_pin_disable_callback(struct device *port, u32_t pin)

There is a few more functions also - but the interesting part is that the have a common
device struct for all HAL functions so that all hardware access is via device structs. So
I2C looks similar:

int i2c_configure(struct device *dev, u32_t dev_config)
int i2c_transfer(struct device *dev, struct i2c_msg *msgs, u8_t num_msgs, u16_t addr)
int i2c_write(struct device *dev, u8_t *buf, u32_t num_bytes, u16_t addr)
int i2c_read(struct device *dev, u8_t *buf, u32_t num_bytes, u16_t addr)

I think it is a nice way to structure HAL API:s as that makes it possible to easily have as many i2c buses or GPIO ports as the platform happens to have without changing anything. They also have mechanisms for init of all devices at different "levels" (similar to our init stages) so that each device declares when it needs to be initialised.

g-oikonomou, 2018-08-16:
Following the discussion earlier on gitter, here is a stab at an updated GPIO HAL, with corresponding adjustments to the LED, SPI HALs and also the generic external flash driver: https://github.com/g-oikonomou/contiki-ng/tree/contrib/gpio/with-ports

* We use `GPIO_HAL_CONF_PORT_PIN_NUMBERING` to configure the HAL to support either port/pin GPIO numbering, or the current simple number.
* Functions have a `gpio_hal_arch_port_foo()` and a `gpio_hal_arch_no_port_foo()` flavour. Platform developers only have to implement one of those two symbols (or to simply provide a macro, or a static inline implementation, same as now).
* Those functions must not be called directly in platform-independent code, cause that would make the code stop being platform-independent. Instead, `gpio_hal_arch_foo(port, pin, ...)` macros used to automatically select the correct function variant. These can be used in a platform-independent fashion (see e.g. how it's done in the updated LED and SPI HAL).
* The port datatype is currently just a number for testing purposes, but it can be typedefed to whatever, I have no strong feelings whatsoever about this. So it can be e.g. a struct. We'll have to change functions to take a pointer argument if we go for a struct.
* Functions that manipulate multiple pins using masks, can still be used only they are limited to pins within a single port. This is exactly how the CC2538 registers operate so I think it should be fine for any other future platform.
* I'll have to adjust the button HAL. Will be a bit tricker and largely-dependent on the datatype chosen for the port representation.

@joakimeriksson looking forward to your thoughts and hopefully that meets your requirements for the port you are working on.

joakimeriksson, 2018-12-28:
One question on this: Is this the correct interpretation of this function:

joakimeriksson, 2018-12-28:
Probably found the better function:
void gpio_hal_arch_port_write_pins(gpio_hal_port_t port,
                                   gpio_hal_pin_mask_t pins,
                                   gpio_hal_pin_mask_t value)
{
  GPIO_PortOutSetVal(port, value, pins);
}

g-oikonomou, 2018-12-28:
> Probably found the better function:

```
void gpio_hal_arch_port_write_pins(gpio_hal_port_t port,
gpio_hal_pin_mask_t pins,
gpio_hal_pin_mask_t value)
{
GPIO_PortOutSetVal(port, value, pins);
}
```

Looks about right!

g-oikonomou, 2019-09-20:
Fixed with #694

closed 2019-09-20 by g-oikonomou