contiki-ng/contiki-ng #564: Add support for TI cc13x2 and cc26x2 platforms

firmwareguru, 2018-05-31:
This issue is to track the progress of the various porting efforts in the Contiki ecosystem to support the TI cc13x2 and cc26x2 devices.  These platforms should be ported into Contiki and Contiki-NG at the same time.

Porting efforts known at this time:

- https://github.com/firmwaremodules/contiki/tree/cc26x2-cc13x2
The goal of this porting effort is to investigate and implement maximum code re-use across all cc13xx and cc26xx device families.
- https://github.com/necropaz/contiki-ng/tree/develop

Relevant Contiki-NG issues:
- #83

firmwareguru, 2018-06-01:
I'm looking at this in the context of Contiki first.  I've started a [wiki ](https://github.com/firmwaremodules/contiki/wiki/Texas-Instruments-CC26X2-CC13X2-Contiki-Port )to document what is happening.

simonduq, 2018-06-02:
That's awesome :)

MartinNPN, 2018-06-05:
Hello,

ti is even working on a port, maybe this would be the cleanest way. I think that they will open a PR before October.

firmwareguru, 2018-09-15:
Alright, looks like it's done!  See PR #609.  I'm closing this.

closed 2018-09-15 by firmwareguru