contiki-ng/contiki-ng #574: The Shell won't build for MAKE_NET_NULLNET

g-oikonomou, 2018-06-07:
It's clearly documented in the wiki that the shell is meant for "interactive IPv6 host inspection and maintenance.", but perhaps we should do something to prevent users from trying to include this in non-v6 firmwares.

Here is what happens with the shell example. For this particular example we can just change the Makefile to force `MAKE_NET_IPV6`, not sure how to handle other examples that simply include the module.

```
$ make MAKE_NET=MAKE_NET_NULLNET
using saved target 'zoul'
  CC        ../../../arch/cpu/cc2538/./ieee-addr.c
  CC        ../../../arch/cpu/cc2538/cc2538.lds
  CC        ../../../arch/cpu/cc2538/./startup-gcc.c
  CC        example.c
  LD        example.elf
obj_zoul/shell-commands.o: In function `cmd_ipaddr':
shell-commands.c:(.text.cmd_ipaddr+0x70): undefined reference to `uip_ds6_if'
obj_zoul/shell-commands.o: In function `cmd_routes':
shell-commands.c:(.text.cmd_routes+0x1e): undefined reference to `uip_ds6_defrt_choose'
shell-commands.c:(.text.cmd_routes+0x22): undefined reference to `uip_ds6_defrt_lookup'
shell-commands.c:(.text.cmd_routes+0x54): undefined reference to `uip_ds6_route_num_routes'
shell-commands.c:(.text.cmd_routes+0x5c): undefined reference to `uip_ds6_route_num_routes'
shell-commands.c:(.text.cmd_routes+0x70): undefined reference to `uip_ds6_route_head'
shell-commands.c:(.text.cmd_routes+0xc6): undefined reference to `uip_ds6_route_nexthop'
shell-commands.c:(.text.cmd_routes+0xe8): undefined reference to `uip_ds6_route_next'
obj_zoul/shell-commands.o: In function `cmd_ping':
shell-commands.c:(.text.cmd_ping+0x34): undefined reference to `uiplib_ip4addrconv'
shell-commands.c:(.text.cmd_ping+0xaa): undefined reference to `uip_icmp6_send'
obj_zoul/shell-commands.o: In function `cmd_ip_neighbors':
shell-commands.c:(.text.cmd_ip_neighbors+0x10): undefined reference to `uip_ds6_nbr_head'
shell-commands.c:(.text.cmd_ip_neighbors+0x54): undefined reference to `uip_ds6_nbr_get_ipaddr'
shell-commands.c:(.text.cmd_ip_neighbors+0x70): undefined reference to `uip_ds6_nbr_get_ll'
shell-commands.c:(.text.cmd_ip_neighbors+0xac): undefined reference to `uip_ds6_nbr_next'
obj_zoul/shell-commands.o: In function `shell_commands_init':
shell-commands.c:(.text.shell_commands_init+0x4): undefined reference to `uip_icmp6_echo_reply_callback_add'
obj_zoul/shell.o: In function `shell_output_6addr':
shell.c:(.text.shell_output_6addr+0xc): undefined reference to `uiplib_ipaddr_snprint'
collect2: error: ld returned 1 exit status
make: *** [example.elf] Error 1
rm obj_zoul/startup-gcc.o example.o
```

simonduq, 2018-06-08:
Agree it would be a good idea to have the shell also in non-IPv6 scenarios. Useful for general system or MAC-related commands.

g-oikonomou, 2018-06-08:
@simonduq please review discussion on gitter with @angelos-oikonomopoulos when you get a mom. We are thinking about splitting commands to respective modules (e.g. TSCH commands in a TSCH module etc)

simonduq, 2018-06-08:
Haven't read the details on how to do it, but feature-wise, this is an absolutely necessary evolution for the shell IMO.

g-oikonomou, 2018-06-08:
I would suggest then:

* TSCH commands in `tsch.c`.
* 6top in `sixtop.c`
* RPL classic/lite commands in `rpl-classic/rpl.c` `rpl-lite/rpl.c`. Not sure where to put those common to both.
* `help`, `reboot` and `log` can stay where they are.
* Not decided what I'd recommend for the other networky ones.

Somewhat confused about this:

```
#if UIP_CONF_IPV6_RPL
  { "rpl-set-root",         cmd_rpl_set_root,         "'> rpl-set-root 0/1 [prefix]': Sets node as root (1) or not (0). A /64 prefix can be optionally specified." },
  { "rpl-local-repair",     cmd_rpl_local_repair,     "'> rpl-local-repair': Triggers a RPL local repair" },
#if ROUTING_CONF_RPL_LITE
  { "rpl-refresh-routes",   cmd_rpl_refresh_routes,   "'> rpl-refresh-routes': Refreshes all routes through a DTSN increment" },
#endif /* ROUTING_CONF_RPL_LITE */
  { "rpl-global-repair",    cmd_rpl_global_repair,    "'> rpl-global-repair': Triggers a RPL global repair" },
#endif /* UIP_CONF_IPV6_RPL */
#if ROUTING_CONF_RPL_LITE
  { "rpl-status",           cmd_rpl_status,           "'> rpl-status': Shows a summary of the current RPL state" },
  { "rpl-nbr",              cmd_rpl_nbr,              "'> rpl-nbr': Shows the RPL neighbor table" },
#endif /* ROUTING_CONF_RPL_LITE */
```

Why are the two in the bottom outside the `UIP_CONF_IPV6_RPL` conditional?

 Also happy with new respective files, e.g. `tsch-commands.c`. Probably a good idea anyway.

simonduq, 2018-06-08:
For the two RPL commands at the bottom, it's simply that their implem is only available for RPL Lite. Agree with additional `-commands.c` files. For things common between RPL Classic and Lite we might have to duplicate. When you think about there's already quite some duplication between the two.

g-oikonomou, 2018-06-08:
> For things common between RPL Classic and Lite we might have to duplicate. When you think about there's already quite some duplication between the two.

Yep, agreed. `rpl/lite` and `rpl/classic` would be really cool. Perhaps one day... :)

simonduq, 2019-01-12:
Original issue is resolved with https://github.com/contiki-ng/contiki-ng/pull/772 but the suggestion to move RPL, TSCH and 6top commands to separate modules is not yet implemented

g-oikonomou, 2019-02-08:
Closing this, since the bug has been fixed.

closed 2019-02-08 by g-oikonomou