contiki-ng/contiki-ng #591: etimer affects local variables

Benedikt1992, 2018-07-05:
While using event timer the values of certain local variables doesn't behave as expected. Which means they contain values that are not reasonable. I couldn't find a pattern which variable will be affected but in a given program it's always the same variable.

In the original use case the etimer is used to poll a value `k` times. Afterwards a timeout happens.

I've build a minimal example based on the hello-world example:
```
#include "contiki.h"

#include <stdio.h> /* For printf() */
/*---------------------------------------------------------------------------*/
PROCESS(hello_world_process, "Hello world process");
AUTOSTART_PROCESSES(&hello_world_process);
/*---------------------------------------------------------------------------*/
PROCESS_THREAD(hello_world_process, ev, data)
{

  PROCESS_BEGIN();

  static struct etimer timer;
  etimer_set(&timer, CLOCK_SECOND);
  int k;
  for (k = 0; k < 10; ++k) {
    printf("Hello, world k: %d\n", k);
    PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&timer));
    etimer_restart(&timer);
  }
  printf("final k: %d\n",k);

  PROCESS_END();
}
```
In this case `k` won't be increased anymore and stays at `1` and the final `printf` is never reached. In the original application  `k` is set to a very huge number (>30.000). I had also situation were completely different variables were affected.

I am pretty sure that this is a bug. Nevertheless, I am new to contiki-ng. Please let me know if I used the timer in an unsupported way.

ivan-alekhin, 2018-07-05:
@Benedikt1992,
I guess the stack variable `k` is getting changed because of the context switch in `PROCESS_WAIT_EVENT_UNTIL`. Basically, you cannot "reuse" stack variables in a code block where you are using a context switch: one of the other processes may have a stack variable in the same place in the memory, and if this process changes its variable, the variable in your process also will be changed.
The simplest solution here is just to make `k` static, i.e.:
`static int k`;
So basically, it's not a bug, it's a well-known feature of a contiki process.
You can find more details [here](https://github.com/contiki-ng/contiki-ng/wiki/Documentation:-Processes-and-events#waiting-for-events). Specifically, consider this comment:
```
Note that this variable must be static in order to preserve the value across yielding.
```

Benedikt1992, 2018-07-05:
Thank you for your reply. Indeed changed `static` the behaviour to the expected one.

closed 2018-07-05 by Benedikt1992