contiki-ng/contiki-ng #602: Testing contiki-ng implementation which includes TSCH and the 6TiSCH with F-interop platform

tamouna, 2018-07-11:
I'm trying to execute the test cases 6TiSCH implemented on F-interop platform.  I choosed contiki-ng implementation for running these tests. I got stuck on the 1st test case "TD_6TiSCH_SYN_01" which aims to check that a 6TiSCH node keep synchronized through pinging it.

I cannot ping the 6TiSCH node from my DAGroot. I flashed the code "simple-node" on my openmote cc2538 to act as a 6TiSCH node.

Here is the Test description of the first test case I want to execute :  
[image]

All the test cases are detailed here : https://openwsn-builder.paris.inria.fr/job/td-6tisch%20builder/lastSuccessfulBuild/artifact/6tisch_plugtest_td_june_2018.pdf 

Can you please help me to execute them by using contiki-ng implementation?

yatch, 2018-07-11:
Can you try `node` under `examples/6tisch/etsi-plugtest-2017`? That is what we used in the plugtest in **2017**. Actually, I'm trying to do almost the same things here with `openmote-cc2538` as you're doing.

tamouna, 2018-07-11:
Do I need to modify the node.c file for executing the first test case?

yatch, 2018-07-11:
I don't think so as long as you can control your board via `shell`.

tamouna, 2018-07-12:
I need to configure my 6TiSCH node to communicate on a single frequency. I'm using examples/6tisch/etsi-plugtest-2017 example.
In the available commands in the shell I didn't find the command I need to type for chosing my channel number.

yatch, 2018-07-12:
@tamouna you're right; to change a channel or the channel hopping sequence, you need to edit `project-conf.h` and re-build your firmware. You may find other relevant settings in `project-conf.h` such as PAN-ID.

tamouna, 2018-07-12:
did you manage to run the first test case?

yatch, 2018-07-12:
No; I had a local issue on using my boards and I haven't had time to look into it...  :-/

tamouna, 2018-07-12:
I modified the ` project-conf.h ` file but my DAGroot and my 6TiSCH node don't synchronized to each other. In wireshark I see that each one send an 'Enhanced Beacon'.

Can you send me the ` project-conf.h ` file modified with the appropriate parameters?

yatch, 2018-07-12:
> I see that each one send an 'Enhanced Beacon'.

If you can see EBs sent by both of the nodes , they are supposed to be synchronized. Otherwise, both of them are working as DAGroot.

Again, I've not run the tests yet...

tamouna, 2018-07-13:
@yatch when will you start to execute the test cases ? I can run them at the same time as you so you can help me to execute them.

joakimeriksson, 2026-10-21:
Closing, since this is a usage question about an external test campaign from 2018. The etsi-plugtest-2017 example (examples/6tisch/etsi-plugtest-2017) is the best starting point. Please use GitHub Discussions for further help.

closed 2026-10-21 by joakimeriksson