contiki-ng/contiki-ng #605: MQTT - QoS 1

manoj6932, 2018-07-17:
Hi, MQTT Client with **QoS Level 1** Seems to having problem in Sending the Publish Ack packet to Broker. Due to this, the Broker send the Message to Node on every 20 Sec and also the Message Identifier is not split from the Payload [@ QoS Level 1]. With QoS level 0,Everything Seems working. Please Find the attached MQTT LOG File
[METER_LOG_asc.txt](https://github.com/contiki-ng/contiki-ng/files/2201437/METER_LOG_asc.txt)

g-oikonomou, 2018-07-17:
Thanks for reporting this, I had a quick look. In [mqtt.c, L844](https://github.com/contiki-ng/contiki-ng/blob/develop/os/net/app-layer/mqtt/mqtt.c#L844) there is this check:

```
  if(((conn->in_packet.fhdr & 0x09) >> 1) > 0) {
    PRINTF("MQTT - Error, got incoming PUBLISH with QoS > 0, not supported atm!\n");
  }
```

This file uses its own debugging approach (we need to change that one day...) hence why it doesn't appear in your logs. I suspect if you change `DEBUG` to 1 you will start seeing this message.

So, in summary, you are not seeing those PUBACKs because our client rejects incoming PUBLISH messages with QoS 1 (or 2).

I didn't spend a lot of time looking at it, but I think adding support could prove tricky. Especially if the PUBLISH is received while the client is in the process of sending a different message over the same connection.

manoj6932, 2018-07-24:
Ok. Thanks

closed 2018-11-01 by g-oikonomou

veracl, 2020-07-16:
Are there any news on this issue?

Side note, perhaps this should be clarified here: https://github.com/contiki-ng/contiki-ng/blob/b896dc871d04cf63d1ade5e1ca1f15f63afa27d7/os/net/app-layer/mqtt/mqtt.h#L38

After reading that description, I expected this to be supported and was surprised that it is not.