contiki-ng/contiki-ng #608: RPL clique reconnection problem

ianno, 2018-07-23:
I'm observing an issue which prevents nodes to reconnect to the network under a particular circumstance. @manoj6932 has reported a similar issue on gitter.
Here is how I think the problem can be replicated:
 
I started from a fresh installation of Contiki-NG v 4.1 (although the problem seems happening also under the develop branch). I'm using the rpl-udp example, modified to use TSCH and Orchestra with default parameters. I'm using 3 ReMOTEs rev b. Then follow the steps:

1. start with
Node A: coordinator, visible by node B
Node B: visible by both A and C
Node C: visible by node B.

2. Network starts, nodes are transmitting and everything is cool. Coordinator shows the routes: B->A; C->B->A.

3. Move B close to C, and make sure that A does not see any other node.

4. After ~5/10 minutes, move node C where B was. This means that:
A sees C
C sees both A and B
B sees C

5. Here is the problem, now the network does not recover, meaning that neither C or B are able to establish a route to A.
Also, moving B closer to A (so that all nodes are visible to each other, as it can be verified by the shell command 'ip-nbr') does not solve the problem.
B and C keep printing: dis_input: not in an instance yet, discard

6. Switching off either B or C, makes the other node disconnect from the network and eventually connect back to A (after a bunch of re-synchronizing on B in case B is off, printed by C)

On Cooja, same scenario, everything works fine.

It seems that B and C keep synchronizing to each other, ignoring A...

manoj6932, 2018-07-24:
The Problem seem to happened by RPL_CONF_DEFAULT_LIFETIME_UNIT and RPL_CONF_DEFAULT_LIFETIME in rpl_conf.h . Adjust the value as per your requirement. [Page 51](https://tools.ietf.org/html/rfc6550) **DODAG Configuration**.

ianno, 2018-07-24:
@manoj6932, unfortunately it seems the problem is not the RPL_DEFAULT_LIFETIME constant. The default value for it is 30  *RPL_DEFAULT_LIFETIME_UNIT = 1800 seconds. In this case, instead, B and C stay disconnected from A way after that limit. Indeed, after that point A does not even list them in its routing table, yet the network does not recover.

ianno, 2018-07-24:
@yatch I think that what you wrote explains why everything works fine in Cooja. 
What I mean by "A sees C", I mean that the two nodes are close enough to be able to communicate directly. No packets are exchanged though.

However, I don't think that changing ```TSCH_DESYNC_THRESHOLD``` will help in this case.
Its default value is 2 minutes, but here the nodes are still disconnected from the coordinator well beyond 60 minutes.

It seems that the two nodes create a loop, keeping synchronizing to each other, thus never leaving the network.
Here's is the result of ```tsch-status``` on both nodes after some time >  ```TSCH_DESYNC_THRESHOLD``` (where their MAC addresses have been replaced with B and C for simplicity):

Node B
```
TSCH status:
-- Is coordinator: 0
-- Is associated: 1
-- PAN ID: 0xabcd
-- Is PAN secured: 0
-- Join priority: 9
-- Time source: C
-- Last synchronized: 0 seconds ago
-- Drift w.r.t. coordinator: -107 ppm
```
Node C:
```
TSCH status:
-- Is coordinator: 0
-- Is associated: 1
-- PAN ID: 0xabcd
-- Is PAN secured: 0
-- Join priority: 7
-- Time source: B
-- Last synchronized: 2 seconds ago
-- Drift w.r.t. coordinator: -114 ppm
```

I think this should not happen, as the network should detect there is a circular dependency and have nodes gradually disconnect and look for the coordinator.

yatch, 2018-07-24:
@ianno Thank you for providing the following information and "TSCH status" dumps, which helped me understand what is happening on your side. 

> Its default value is 2 minutes, but here the nodes are still disconnected from the coordinator well beyond 60 minutes.

ianno, 2018-07-24:
Of course. Thanks for helping!

manoj6932, 2018-07-26:
@ianno first of all i'm sorry, if anything wrong in my points because I'm new to contiki. I saw the changes in the RPL behavior by changing the Default Lifetime Value.

ianno, 2018-07-26:
Oh, no need to be sorry, @manoj6932 :-).
Having a conversation going is always a good thing. Your comment might help someone else.

yatch, 2018-08-01:
@ianno The root cause of this is "synchronization loop". 

Disassociation or resetting the MAC could solves this issue; but it seems costly. I would propose another.

One naive solution is to remove the time source when a node loses its preferred parent. This could be done by modifying `tsch_rpl_callback_parent_switch` so that the function calls `tsch_queue_update_time_source()` with `NULL` when `new` (the new preferred parent) is `NULL`. Also, it should stop sending EBs. Then, nodes get disassociated eventually, I think (I've not tested it yet). This would give the nodes some chance to receive a valid EB or DIO which get them back to the network.

By the way, removing the time source is prevented on purpose as per the comment found just above `tsch_queue_update_time_source()`.

I'd like to have more time to look into this issue.

ianno, 2018-08-01:
@yatch thanks for looking into this.

> The root cause of this is "synchronization loop".
yep, it seems so.

> Disassociation or resetting the MAC could solves this issue; but it seems costly.
...
> One naive solution is to remove the time source when a node loses its preferred parent.[...] Then, nodes get disassociated eventually
Isn't the end result the same, though? Removing the time source when there is no preferred parent, so that eventually the node disconnects, seems pretty much equivalent, if not more twisted, than just disassociate when we leave the DAG. Additionally, there might be the risk to have the node find yet another incorrect time source after we remove the old time source but before it disassociate.

Anyway, I've been testing further the (temporary?) solutions proposed [above](https://github.com/contiki-ng/contiki-ng/issues/608#issuecomment-409319470) and they seem working. 
But I do agree that a more careful analysis is needed. 
Also because this bug is potentially disruptive in real-world scenarios. I.e., a temporary isolation of the root, or even just rebooting it, could cause the rest of the network to end up in a synchronization loop and it would require a manual reboot of all the other nodes.

I'll keep you posted in case something new comes up.

debug-ito, 2018-11-22:
A possible workaround for this problem is to suppress transmission of Enhanced Beacons until the node fully joins to the RPL DODAG. This is described in section 6.3 of RFC 8180, and implemented in p-r #769.

I'm not sure if this is a theoretically decisive solution. However, we found in experiments with real devices that this workaround successfully eliminated TSCH synchronization loops. In the experiments the nodes were fixed, so I'm not sure if it works for mobile nodes.

debug-ito, 2018-12-07:
I've been considering the problem, and concluded p-r #769 partly solves this problem.

I think the following events can happen between step 3 and step 4.

- (3-1) B detects A is not reachable anymore.
- (3-2) B starts poisoning (i.e. sending DIO with +∞ rank)
- (3-3) C receives the DIO from B, and it starts poisoning, too.
- (3-4) B times out in synchronization to A. B leaves TSCH PAN and starts scanning EBs.
  However, C is still sending EBs for a short period of time because it thinks B is still in the PAN and it synchronizes to B.
- (3-5) B catches an EB from C. B synchronizes to C. Now a TSCH synchronization loop is formed.

P-r #769 prevents a node from sending EBs until it joins to the RPL DODAG. With this enabled, C stops sending EBs at step (3-3), so B won't catch them at step (3-5).

However, I can think of another scenario.

- (3-3') B sends TSCH keep-alive to A, but it fails.
- (3-4') B sets the time source to C (in `resynchronize` function in tsch.c). It sends keep-alive to C, and it succeeds.

In this case, p-r #769 is powerless.

So, I propose to choose the TSCH time source **always from RPL parent set** (as stated in section 6.2, RFC 8180). If the parent set is empty, it should just leave TSCH PAN and start scanning for EBs. This is practically a little modified version of [ianno's proposal above](https://github.com/contiki-ng/contiki-ng/issues/608#issuecomment-409319470).

With my proposal enabled, B would not set the time source to C at step (3-4'). It would leave the TSCH PAN instead.

ianno, 2018-12-07:
That's interesting.
I'm going to test this approach soon.

PR #1166 (2020-03-06): keep TSCH timesource state consistent with RPL parent state
Fixes #890 and #877

Potentially fixes #608
closed 2020-03-06 by simonduq