contiki-ng/contiki-ng #616: A Makefile within a `MODULES_REL` dir cannot override (some) build variables

g-oikonomou, 2018-08-11:
Imagine a platform `Makefile` that does this:

```
SOME_MAKE_VARIABLE ?= some-default-value
```

And then we have an example with sub-dirs. Those subdirs are included through `MODULES_REL`. One of those subdirs has a `Makefile` which tries to do this:

```
SOME_MAKE_VARIABLE = some-example-specific-value
```

This will fail, since this example `Makefile` will get included after the platform-specific one. The final value of `SOME_MAKE_VARIABLE` will be `some-default-value`, which IMHO is not what we want.

g-oikonomou, 2018-08-12:
I've not spent time to figure that out exactly, the assignment in the MODULE makefile seems to be getting ignored altogether :S

g-oikonomou, 2018-08-12:
Hmm perhaps I need more caffeine here!

simonduq, 2018-08-12:
Yes, I see exactly what you're referring to. I'm not sure if we can do much better with the limitations of `make`.

closed 2019-04-19 by g-oikonomou