contiki-ng/contiki-ng #619: Global ip-addr removed if root is rebooted, rejoins network, and then starts acting as root again

jeppenodgaard, 2018-08-14:
The root seems to have an issue when rebooted in rpl-classic, when other nodes are not rebooted.
The issue can be reproduces using examples/multicast (release/v4.1) in the Cooja simulator:

1. Delete line examples/multicast/root.c:114
2. Add 'MODULES += os/services/shell' to examples/multicast/Makefile
3. Create 3 sink.c nodes with mote ids 1-3.
4. Create 1 root.c node with mote id 4, which is the last mote id (IMPORTANT).
5. Place the motes in a line: root <-> sink <-> sink <-> sink.
6. Start the simulation, send via shell 'rpl-set-root 1' to root and wait for the network to build.
7. Send 'ip-addr' via shell, which should show:
    Node IPv6 addresses:
    -- fd00::204:4:4:4
    -- fe80::204:4:4:4
8. Delete the root mote.
9. Add the root mote again with same position and make sure it is still mote id 4.
10. Start the simulation and keep sending 'ip-addr' via shell until it prints:
    Node IPv6 addresses:
    -- fd00::204:4:4:4
    -- fe80::204:4:4:4
11. Send 'rpl-set-root 1' via shell, and then send 'ip-addr' via shell:
    Node IPv6 addresses:
    -- fe80::204:4:4:4

The address fd00::204:4:4:4 never returns to the ip-addr list, and it is also possible to keep sending 'rpl-set-root 1' via shell.

simonduq, 2018-08-15:
Sounds like the root joins its old network first, and then something goes nuts when it attempts to become root. Thanks for reporting.

jeppenodgaard, 2018-10-22:
I've opened a pull request to attempt to fix the issue:
#727

PR #727 (2018-11-15): Fix: Global ip-addr removed if root is rebooted, rejoins network, and then starts acting as root again
Added test to reproduce issue, and proposed fix in rpl-dag, where dag is only freed if dag_id is different.

Fixes #619