contiki-ng/contiki-ng #625: TSCH backup link

seungbumz, 2018-08-22:
Hi. I am studying TSCH protocol.

In TSCH, backup link is used when there is no packet to send in current link.

And the comment in the code says "Backup link must have Rx flag".

It means, the link with Tx flag only (without Rx flag) is not available for backup link.

Why??

seungbumz, 2018-08-22:
> Hi,
>
> The backup link is an implementation link, not intended to stay in the
> long run.
> The problem was that selecting a link actually can take significant
> processing time, depending on the schedule. This can affect timeslot
> timings. For this reason, we select links ahead of time, that is, right
> after a given link, we select the next. But we don't know yet what the
> queues will look like when the time of the link comes.
> In principle you could have any number of backup links, with or without
> Rx. The good thing with having an Rx flag is that you know for sure you'll
> be able to actually run the link, regardless of Tx queues.

Thanks for your reply.

It will be very helpful to my work.

firmwareguru, 2018-09-15:
@seungbumz Can we close this issue then?

closed 2018-09-15 by seungbumz