contiki-ng/contiki-ng #628: multiple definition of `ecc_ecdsa_sign' when targeting srf06-cc26xx with dtls

tdesmet, 2018-08-23:
If you build the coap-example-client for the cc1310 launchpad with DTLS you get the following error:
`make TARGET=srf06-cc26xx BOARD=launchpad/cc1310 MAKE_WITH_DTLS=1`
```
obj_srf06-cc26xx/ecc.o: In function `ecc_ecdsa_sign':
ecc.c:(.text.ecc_ecdsa_sign+0x0): multiple definition of `ecc_ecdsa_sign'
obj_srf06-cc26xx/rom_crypto.o:(.data.ecc_ecdsa_sign+0x0): first defined here
collect2: error: ld returned 1 exit status
```

g-oikonomou, 2018-08-23:
Yep, seen that one mentioned before. Clearly we need to do something about this, either rename one or the other...

tdesmet, 2018-08-23:
any preference on which one to rename and what the new name should be?

g-oikonomou, 2018-08-23:
Selfishly, I'd rename the one that ships with DTLS :)

tdesmet, 2018-09-12:
I think this issue can be closed?

closed 2018-09-12 by g-oikonomou