contiki-ng/contiki-ng #644: CCA_ENABLED in TSCH

seungbumz, 2018-09-06:
This is the code in tsch-slot-operation.c

/***********************************************CODE*****************************************************/
#if CCA_ENABLED
      static uint8_t cca_status;
#endif
...
...
...

#if CCA_ENABLED
        cca_status = 1;
        /* delay before CCA */
        TSCH_SCHEDULE_AND_YIELD(pt, t, current_slot_start, TS_CCA_OFFSET, "cca");
        TSCH_DEBUG_TX_EVENT();
        tsch_radio_on(TSCH_RADIO_CMD_ON_WITHIN_TIMESLOT);
        /* CCA */
        BUSYWAIT_UNTIL_ABS(!(cca_status |= NETSTACK_RADIO.channel_clear()),
                           current_slot_start, TS_CCA_OFFSET + TS_CCA);
        TSCH_DEBUG_TX_EVENT();
        /* there is not enough time to turn radio off */
        /*  NETSTACK_RADIO.off(); */
        if(cca_status == 0) {
          mac_tx_status = MAC_TX_COLLISION;
        } else
#endif /* CCA_ENABLED */

#define BUSYWAIT_UNTIL_ABS(cond, t0, offset) \
  while(!(cond) && RTIMER_CLOCK_LT(RTIMER_NOW(), (t0) + (offset))) ;
#endif
/***********************************************END*****************************************************/        

Is it possible **cca_status** is set to 0?

It is initialized to 1,

Regardless of result of NETSTACK_RADIO.channel_clear(), it always 1 by OR operator.

simonduq, 2018-09-06:
Sounds like a bug indeed. This feature is disabled by default and received very little testing (I for one never ever tried it). Bugfix welcome :)

simonduq, 2018-09-15:
Re-opening as this was not fixed as far as I know

PR #681 (2018-10-11): TSCH CCA
The TSCH CCA code has been un-maintained for years. This gets it back to a compiling state, and also fixed #644.
closed 2018-10-11 by g-oikonomou