contiki-ng/contiki-ng #654: RPL_DAG_LIFETIME is set to 60 hours instead of one hour in rpl lite

tdesmet, 2018-09-12:
rpl lite RPL_DAG_LIFETIME is supposed to be set to one hour `#define RPL_DAG_LIFETIME (60 * 60) /* one hour */` but the comments state that it is a multiple of the lifetime unit which is defined as `#define RPL_DEFAULT_LIFETIME_UNIT 60` this makes the RPL_DAG_LIFETIME 60 hours

abhiverma866, 2018-09-27:
* Default route lifetime unit. This is the granularity of time used in RPL lifetime values, in seconds.
Thus, 60*60=3600 seconds i.e 1 hour.

simonduq, 2018-09-27:
But it's in lifetime units, not in seconds

abhiverma866, 2018-09-27:
I guess the code must be modified to..

/* Maximum lifetime of a DAG as a multiple of the lifetime unit. */
#ifdef RPL_CONF_DAG_LIFETIME
#define RPL_DAG_LIFETIME                RPL_CONF_DAG_LIFETIME
#else
#define RPL_DAG_LIFETIME                (60 * 60 * CLOCK_SECOND) /* one hour */
#endif /* RPL_CONF_DAG_LIFETIME */