contiki-ng/contiki-ng #656: The problem of SCA(sleep clock accuracy) in ble_hal_cc26xx.c

howard1985, 2018-09-15:
Hi,
I have one question, when I add gatt/att function based on contiki system, it can not connect too long with mobile APP. I find the function "connection_event_slave" does not think about the dynamic SCA(sleep clock accuracy). For example the ios is 31--50ppm. some android mobile is 251--500ppm. Am I right?

firmwareguru, 2018-09-15:
How can we reproduce this issue? How have you added the BLE GATT profiles?

howard1985, 2018-09-17:
According to our team's many tests, this is a must-have problem. We have a slave device which run contki system, the connection duration depends on the accuracy of the mobile terminal clock.
  It is not very difficult to add the GATT profiles if you know some BLE protocol. It is judged by the frame_type according to the frame_type whether it is a control message (0x03) or a data message (0x02) in the link layer, maybe l2cap layer is best. At this point, you need to read the low-power Bluetooth protocol document. And Write code based on standard processes such as service discovery and feature acquisition.
   Our team's Bluetooth protocol stack implements GATT, part of the ATT function, and can do unpaired connection and communication with the mobile client. Because of Unclear stratification, we are now in reconstructing it.  In the future, we hope it will be uploaded to the Github review.

howard1985, 2018-09-17:
If you have two device, which are master and slave, run the contiki_ng system,  and they must be have differet sleep clock accruracy,  at last, the master device have to handle the connection event by fixed internal time, not like the original file( restart timer after receive the data from slave). I suspect that the phone ble protocol uses the same method. I suppose it alse can reproduce it .( I do not test it )
   Of course,  if your slave device has part of GATT and ATT function, Then, it can connect with some BLE debug app in different mobile(ios and android system). the problem will be reproduce easyly(I just do it).
   so , it have to increase clock synchronization between master and slave.

firmwareguru, 2018-09-18:
Thanks for the additional info @howard1985.  Finding a way to get G/ATT working on top of the existing BLEach/l2cap layer in Contiki-NG is something I'm interested in :)  I guess what you're saying is that once I find a way to do that, it won't work reliably!

firmwareguru, 2018-09-18:
@spoerk Maybe you could weigh in here?

howard1985, 2018-09-28:
Thanks for your reply.
Well, I am sorry. I mean the total ble data communiation duration is short, at first, the connection event duration is ok.  But, it just stay connected for a few minutes with android smartphone, then end by master device because of timeout.
And as you say, SCA is the root problem in every connect enent.
As the time error (because of SCA)of each connection event accumulates, the communication packet loss increases gradually . Until a certain moment, the connection event is completely unable to be synchronized, and the communication  is terminated because of timeout.
By the way, the connection parameters will be set by master device first(connection request command), and master device can change them when it needed. And according to ble protocol, the slave can also 
initiate a request to change connection parameters from l2cap layer.

firmwareguru, 2018-10-05:
Wondering if anyone has considered applying the Mynewt NimBLE (https://mynewt.apache.org/pages/ble/)  stack to Contiki-NG, or at least the ATT/GATT layers.