contiki-ng/contiki-ng #679: CC2650 Launchpad flashing procedure

rienafairefr, 2018-09-28:
Hi, 

I wanted to try out the IPV6 BLE mesh with Contiki-NG using CC2650 boards we have around, and also prepare to deploy the boards in [FIT IoT-LAB](www.iot-lab.info), but I can't get the board to flash, and the doc on the [wiki](https://github.com/contiki-ng/contiki-ng/wiki/Platform-srf06-cc26xx#over-jtag) is not getting me to the solution.

Hardware situation:
- Brand-new CC2650 Launchpad Rev 1.3 w Firmware 2.2.5.1
Software situation:
- Connecting from a Linux laptop, but I will need to be able to flash from an ARM Single Board Computer

Tried:
- using Uniflash 4.4 for Linux, in the GUI I get "Cannot read property 'DS' of undefined"
- using Uniflash 4.4 CLI (dslite.sh, as is done inside RIOT, AFAIK) works fine, doing `dslite.sh --config RIOT/boards/cc2650-launchpad/dist/cc26x0f128_XDS110.ccxml <elf_file>`
The problem with Uniflash is closed source, and impossibility to use it on an ARM based Linux

- using Contiki-NG `.upload` target, (which uses cc2538-bsl which is good) but it requires something about bootloader state, because I get the error described in the docs
```
Reading data from cc26xx-demo.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
../../../arch/cpu/cc26xx-cc13xx/Makefile.cc26xx-cc13xx:91: recipe for target 'cc26xx-demo.upload' failed
```

In the end, I could flash using this specific steps :
- Uniflash/dslite.sh with cc26x0f128_XDS110.ccxml a different firmware than the factory one
- resetting by holding the buttons on the side, then
- `PORT=/dev/ttyACM0 BOARD=launchpad/cc2650 make cc26xx-demo.upload`

What can I do/flash on the board so that I can just plug the board then directly flash with cc2538-bsl ? Erasing the flash works, but erasing the flash can only be done with uniflash, which I won't have on the Single Board Computer.

I confirmed the above, I can also do
- Uniflash erase using `dslite.sh --mode flash --config cc26x0f128_XDS110.ccxml -O PinReset --before Erase`
- `PORT=/dev/ttyACM0 BOARD=launchpad/cc2650 make cc26xx-demo.upload`

Any help is appreciated, pinging @g-oikonomou because he wrote the docs on srf06 cc26xx

g-oikonomou, 2018-09-29:
From what you are describing, everything seems to be behaving as expected, other than some problems getting uniflash to work on Linux.

On the topic of programming from the single board: We have done something similar in the past to programme devices remotely using RPis.

What you can do is connect two RPi GPIO pins to a GPIO and the RESET pins on the launchpad. Write a little script on the RPi that pulls down (or up) the GPIO and then toggles RESET. This will put the Launchpad in bootloader mode and then you can emit the cc2538-bsl command. You will need to configure the launchpad code to enter bootloader mode when the desired pin is high / low - see `ccxxware-conf.h`

rienafairefr, 2018-10-01:
Thanks @g-oikonomou 

Yeah, I think I'll settle for something using RPi GPIO, should work. 
If I understand correctly, to have the chip boot in bootloader mode, the current firmware on the flash should be configured a certain way (with the CCXXWARE_CONF_BL_LEVEL and CCXXWARE_CONF_BL_PIN_NUMBER), which are burned into the chip in the `.ccfg` page. This might be problematic: I wouldn't be in 100% control of the flashed firmware, users would decide what to flash, and they could flash something with the .ccfg differently setup. I might have to hand-modify the binary/hex they want to flash to ensure that the config is right, but I'm disgressing and thinking out loud here.

I've also encountered the SimpleLink OpenOCD on [TI's website](http://www.ti.com/tool/SIMPLELINK-OPENOCD), and it's apparently been [upstreamed](http://openocd.zylin.com/#/c/4358/) somewhat, have you ever tried it ?

g-oikonomou, 2018-10-01:
You understand correctly.

One thing you can do is, make sure to not overwrite CCFG. So for example, you can limit user firmware to not use the last flash page. Or you can make sure to write your CCFG instead of the one linked in firmware. Modifying user firmware will be very straightforward with srecord.

rienafairefr, 2018-10-01:
Re TI's OpenOCD : 

TI's openocd modifications seem to do the trick, flash/erase/reset working seemingly OK, without putting the chip in bootloader mode using the DIO, using latest code dated September 2018 commit 4765fd4c864c47d48754350c196bd5c5ce9c5ed9 in https://git.ti.com/sdo-emu/openocd  . This will probably be upstreamed at some point and put in openocd 0.11 
Not sure which strategy (DIO-bootloader-mode with regular cc2538-bsl or openocd) we'll use in the end, but thanks @g-oikonomou this has been an interesting discussion

closed 2018-10-13 by g-oikonomou