contiki-ng/contiki-ng #696: PLATFORMS_EXCLUDE and friends can be tricked

g-oikonomou, 2018-10-13:
From `examples/multicast`:

```
$ make TARGET=sky
Skipping root: not for the 'sky/' platform!
Skipping intermediate: not for the 'sky/' platform!
Skipping sink: not for the 'sky/' platform!

$ make sink TARGET=sky
Skipping sink: not for the 'sky/' platform!
```

But:

```
$ make sink.sky TARGET=sky 
mkdir obj_sky
  CC        ../../os/contiki-main.c
  CC        ../../arch/platform/sky/./contiki-sky-platform.c
[...]
  AR        contiki-ng-sky.a
  CC        sink.c
  LD        sink.sky
rm sink.o
```

I think we need something like this (untested):
```
$ git diff
diff --git a/Makefile.include b/Makefile.include
index 319ae29..597d42d 100644
--- a/Makefile.include
+++ b/Makefile.include
@@ -483,7 +483,7 @@ endif
 
 ifeq ($(PLATFORM_ACTION),skip)
 # Skip this target.
-$(CONTIKI_PROJECT):
+%.$(TARGET):
        @echo "Skipping $@: not for the '$(TARGET)/$(BOARD)' platform!"
 else
 # Build this target.
```

atiselsts, 2018-10-15:
Not sure that this is an issue? By default, the Makefile will stop anyone trying to build things for inappropriate platform. If someone knows what he's doing and has a reason (for example, testing after porting things to a new platform), they can override this default behavior by explicitly telling to build `sink.xxx` for the xxx platform.

Is there a problem you're trying to solve?

simonduq, 2018-10-15:
I would agree with @g-oikonomou , even if only for the sake of consistency, in the example above, I'd want `make sink.sky TARGET=sky` and `make TARGET=sky` to have the same behavior.

g-oikonomou, 2018-10-15:
The worst problem I can think of is someone accidentally (or intentionally) triggering a build, getting a linker overflow error and then opening a github issue about it :)

simonduq, 2018-11-21:
OK here https://github.com/contiki-ng/contiki-ng/issues/766 is one more reason the issue here must be resolved

g-oikonomou, 2019-02-08:
#806

closed 2019-02-08 by g-oikonomou