contiki-ng/contiki-ng #719: renode not building correctly during docker image updates

g-oikonomou, 2018-10-19:
```
Done building project "/home/user/renode/Renode.sln".
Build succeeded.
	 0 Warning(s)
	 0 Error(s)
Time Elapsed 00:02:41.2492350
cp: cannot stat 'src/Infrastructure/src/Emulator/LLVMDisassembler/bin/Release/libLLVM.*': No such file or directory
The command '/bin/sh -c git clone --quiet https://github.com/renode/renode.git   && cd ${HOME}/renode   && git checkout v1.3   && ./build.sh' returned a non-zero code: 1
Failed to build Docker image
```

Something fishy here. Debugging.

simonduq, 2018-10-20:
@PiotrZierhoffer any clue on this?

g-oikonomou, 2018-10-20:
I also tried to build the docker image locally macos, but I am observing different behaviour. During the build I am seeing multiple:
```
<path>/<proj>.cproj: error : Error executing task CCompilerTask: One or more errors occurred.
```

and then the build ends with:
```
    /home/user/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj: error : Error executing task CCompilerTask: One or more errors occurred.

    0 Warning(s)
    5 Error(s)

    Time Elapsed 00:00:38.0582790
The command '/bin/sh -c git clone --quiet https://github.com/renode/renode.git   && cd ${HOME}/renode   && git checkout v1.3   && ./build.sh' returned a non-zero code: 1
```

g-oikonomou, 2018-10-20:
Sorry for the spam here. Here is a successful old image build https://api.travis-ci.org/v3/job/390233665/log.txt:

```
		Project "/home/user/renode/src/Infrastructure/src/Emulator/LLVMDisassembler/LLVMDisassembler.csproj" (default target(s)):
[...]
			Target DeployOutputFiles:
				Copying file from '/home/user/renode/src/Infrastructure/src/Emulator/LLVMDisassembler/obj/Release/LLVMDisassembler.pdb' to '/home/user/renode/src/Infrastructure/src/Emulator/LLVMDisassembler/bin/Release/LLVMDisassembler.pdb'
				Copying file from '/home/user/renode/src/Infrastructure/src/Emulator/LLVMDisassembler/obj/Release/LLVMDisassembler.dll' to '/home/user/renode/src/Infrastructure/src/Emulator/LLVMDisassembler/bin/Release/LLVMDisassembler.dll'
			Target CopySharedLibraries:
				Copying file from '/home/user/renode/lib/resources/llvm/32_libLLVM-3.4.so' to '/home/user/renode/src/Infrastructure/src/Emulator/LLVMDisassembler/bin/Release//libLLVM.so'
		Done building project "/home/user/renode/src/Infrastructure/src/Emulator/LLVMDisassembler/LLVMDisassembler.csproj".
```

Here is what happens now:

```
Project "/home/user/renode/src/Infrastructure/src/Emulator/LLVMDisassembler/LLVMDisassembler.csproj" (default target(s)):
[...]
			Target DeployOutputFiles:
				Copying file from '/home/user/renode/src/Infrastructure/src/Emulator/LLVMDisassembler/obj/Release/LLVMDisassembler.pdb' to '/home/user/renode/src/Infrastructure/src/Emulator/LLVMDisassembler/bin/Release/LLVMDisassembler.pdb'
				Copying file from '/home/user/renode/src/Infrastructure/src/Emulator/LLVMDisassembler/obj/Release/LLVMDisassembler.dll' to '/home/user/renode/src/Infrastructure/src/Emulator/LLVMDisassembler/bin/Release/LLVMDisassembler.dll'
		Done building project "/home/user/renode/src/Infrastructure/src/Emulator/LLVMDisassembler/LLVMDisassembler.csproj".
```

So it looks like the cp fails because the file libLLVM.so is being skipped for some reason.

As far as I can see the renode git version and all submodule versions are identical between the successful old build and now, so I am kinda stuck here.

PiotrZierhoffer, 2018-10-22:
Ok, I have traced the issue down and I have two possible solutions for you.

I noticed you use a 32-bit version of Ubuntu. We dropped a 32-bit support quite some time ago. This lead us to remove some of the 32-bit binary libraries from the repository we fetch during the build process.

We could re-upload the single file you're missing here, but to move forward with Renode versions you'd need to have a 64-bit image. I altered the Dockerfile a little and got it to work, but the question is if it's acceptable for you to change the image type?

If yes, then we can a) create a merge request with the updated 64bit Dockerfile b) create a merge request updating Renode scripts you host to the newest version (there were a few API changes meanwhile).

If that's unacceptable for you, let me know and we'll re-upload the LLVM binary.

g-oikonomou, 2018-10-22:
Hi @PiotrZierhoffer thanks for investigating. Unfortunately switching to 64-bit would mean losing key cooja functionality. We are trying to address this but until that's done we are stuck with 32-bits, which is very frustrating...

PiotrZierhoffer, 2018-10-22:
I got the image to build, it should work for you as well - we brought the binary back.

I ran a HEAD contiki-ng with the rpl-udp sample and everything seems to work fine.

Unfortunately, with the current version of the Docker image, you won't be able to update Renode. It would be great if you could give us a nudge if you manage to switch to 64bit.

@g-oikonomou could you please give me more info about your macOS problems?

g-oikonomou, 2018-10-22:
I re-triggered #720 and it built renode successfully. Many thanks. Regarding the macOS problems, basically we cannot make Cooja motes to work on 64-bit platforms. This happens in 64-bit linux as well, not just macOS.

I am closing this as fixed, thanks for your support!

simonduq, 2018-10-23:
Thanks a million @PiotrZierhoffer !

PiotrZierhoffer, 2018-10-23:
@g-oikonomou I was referring to this comment: https://github.com/contiki-ng/contiki-ng/issues/719#issuecomment-431593497 and I understand it was Renode failing somehow.

Does it still happen? Just curious, it would be very helpful if you could drop me the log.

g-oikonomou, 2018-10-23:
Ah 

Gocha, let me give it a try again.

g-oikonomou, 2018-10-23:
Typical I cannot reproduce it now, perhaps I was trying with a wrong version of something or other (I did try to use an older version of mono at some stage). The entire image just built fine on macOS

closed 2018-10-23 by g-oikonomou

PiotrZierhoffer, 2018-10-24:
Great, thanks for checking that!