contiki-ng/contiki-ng #722: Docker macos wiki page

g-oikonomou, 2018-10-19:
I am following the instructions for [docker for macos](https://github.com/contiki-ng/contiki-ng/wiki/Docker#on-macos) and I am not getting very far. I can either not start the instance at all, or the `bash -c` command fails.

@yatch I think it was you who wrote this? If you have time to review that would be great. Also, the image is now at `contiker/contiki-ng`, no longer under `simonduq`.

yatch, 2018-10-20:
@g-oikonomou Yes, I wrote this section. And, I'll review and update it. Thanks!

yatch, 2018-10-23:
@g-oikonomou I've update the wiki. I guess you had the following error when you tried to run `bash`:
```
$ contiker bash
cp: missing file operand
Try 'cp --help' for more information.
```
This should be resolved by the update. I tested with both of `Docker for Mac` and `Docker Toolbox on macOS`. I overlooked the errors which don't happen on `zsh` somehow. Thank you for the report.

It'd be great if you can try the updated version of `contiker()`.

g-oikonomou, 2018-10-23:
Thanks @yatch that's exactly the error I was getting.

Now I am running contiker and nothing happens (although it does seem to spend a few seconds thinking about it)

```
$ contiker
$ 
```

By the way, what do you mean by `$ open a xquartz`? Typo?

g-oikonomou, 2018-10-23:
```
$ docker version
Client:
 Version:           18.06.1-ce
 API version:       1.38
 Go version:        go1.10.3
 Git commit:        e68fc7a
 Built:             Tue Aug 21 17:21:31 2018
 OS/Arch:           darwin/amd64
 Experimental:      false

Server:
 Engine:
  Version:          18.06.1-ce
  API version:      1.38 (minimum version 1.12)
  Go version:       go1.10.3
  Git commit:       e68fc7a
  Built:            Tue Aug 21 17:29:02 2018
  OS/Arch:          linux/amd64
  Experimental:     true
```

yatch, 2018-10-23:
Yes! I fixed the typo. And, in order to invoke `bash` in the container, `$ contiker bash` is a right command. In the wiki page, `bash` in the command is missing. This one is also resolved.

In addition, I've added text for 'Without XQuartz' case: https://github.com/contiki-ng/contiki-ng/wiki/Docker#without-xquartz

I'm closing this issue. Thank you!

g-oikonomou, 2019-03-30:
I'm still struggling with Cooja on macOS through docker.

* XQuartz running and accepting incoming connections
* socat running:
```
socat TCP-LISTEN:6000,reuseaddr,fork,range=127.0.0.1/32 UNIX-CLIENT:/tmp/.X11-unix/X0
```

`contiker` alias set up in `.bash_profile` as per instructions. I can fire up the docker container no problems and then:

```
$ cd tools/cooja
$ ant run
[...]
run:
     [java] Invalid MIT-MAGIC-COOKIE-1 keyException in thread "main" java.awt.AWTError: Can't connect to X11 window server using 'host.docker.internal:0.0' as the value of the DISPLAY variable.
     [java] 	at sun.awt.X11GraphicsEnvironment.initDisplay(Native Method)
     [java] 	at sun.awt.X11GraphicsEnvironment.access$200(X11GraphicsEnvironment.java:65)
     [java] 	at sun.awt.X11GraphicsEnvironment$1.run(X11GraphicsEnvironment.java:115)
     [java] 	at java.security.AccessController.doPrivileged(Native Method)
     [java] 	at sun.awt.X11GraphicsEnvironment.<clinit>(X11GraphicsEnvironment.java:74)
     [java] 	at java.lang.Class.forName0(Native Method)
     [java] 	at java.lang.Class.forName(Class.java:264)
     [java] 	at java.awt.GraphicsEnvironment.createGE(GraphicsEnvironment.java:103)
     [java] 	at java.awt.GraphicsEnvironment.getLocalGraphicsEnvironment(GraphicsEnvironment.java:82)
     [java] 	at sun.awt.X11.XToolkit.<clinit>(XToolkit.java:126)
     [java] 	at java.lang.Class.forName0(Native Method)
     [java] 	at java.lang.Class.forName(Class.java:264)
     [java] 	at java.awt.Toolkit$2.run(Toolkit.java:860)
     [java] 	at java.awt.Toolkit$2.run(Toolkit.java:855)
     [java] 	at java.security.AccessController.doPrivileged(Native Method)
     [java] 	at java.awt.Toolkit.getDefaultToolkit(Toolkit.java:854)
     [java] 	at sun.swing.SwingUtilities2.getSystemMnemonicKeyMask(SwingUtilities2.java:2020)
     [java] 	at javax.swing.plaf.basic.BasicLookAndFeel.initComponentDefaults(BasicLookAndFeel.java:1158)
     [java] 	at javax.swing.plaf.metal.MetalLookAndFeel.initComponentDefaults(MetalLookAndFeel.java:431)
     [java] 	at javax.swing.plaf.basic.BasicLookAndFeel.getDefaults(BasicLookAndFeel.java:148)
     [java] 	at javax.swing.plaf.metal.MetalLookAndFeel.getDefaults(MetalLookAndFeel.java:1577)
     [java] 	at javax.swing.UIManager.setLookAndFeel(UIManager.java:539)
     [java] 	at javax.swing.UIManager.setLookAndFeel(UIManager.java:579)
     [java] 	at javax.swing.UIManager.initializeDefaultLAF(UIManager.java:1349)
     [java] 	at javax.swing.UIManager.initialize(UIManager.java:1459)
     [java] 	at javax.swing.UIManager.maybeInitialize(UIManager.java:1426)
     [java] 	at javax.swing.UIManager.getInstalledLookAndFeels(UIManager.java:419)
     [java] 	at org.contikios.cooja.Cooja.setLookAndFeel(Cooja.java:1239)
     [java] 	at org.contikios.cooja.Cooja.main(Cooja.java:3168)
     [java] Java Result: 1

BUILD SUCCESSFUL
Total time: 30 seconds

```

Any clues?

simonduq, 2019-03-30:
Have you seen this?: https://github.com/contiki-ng/contiki-ng/issues/798
I use:
```
XAUTH_HEXKEY=`xauth list | grep smbp.local/unix | awk '{print $3}'`
```

g-oikonomou, 2019-03-30:
I had not seen this, no

closed 2019-03-30 by g-oikonomou

simonduq, 2019-03-30:
Sounds like it helped then :) We should think about updating the wiki page