contiki-ng/contiki-ng #728: Toolchain installation instructions in wiki page

khongyew, 2018-10-22:
The instruction to install the ARM toolchain is current as follow:
`$ sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa
$ sudo apt update
$ sudo apt install srecord gcc-arm-embedded`
This installs version 7 of the toolchain, which failed to compile cc26xx-web-demo using `make TARGET=srf06-cc26xx BOARD=sensortag/cc2650 `

Version 5 of the toolchain works.
instruction for version 5 installation can be found at: [https://github.com/contiki-ng/contiki-ng/blob/develop/tools/vagrant/bootstrap.sh](url)

atiselsts, 2018-10-22:
I agree that relying on `apt install` can give bad results. We've also seen a problem (discovered by an unlucky student) with a version of `arm-gcc` that can successfully build executables, but they do not work when flashed to the node.

g-oikonomou, 2018-10-22:
Yeah these instructions are bad. I'll update all relevant pages, thanks for reporting this.

That's my recommended page to download from: https://developer.arm.com/open-source/gnu-toolchain/gnu-rm

khongyew, 2018-10-22:
Do you guys also use arm-gcc version 5?

g-oikonomou, 2018-10-22:
I have multiple versions installed in parallel and I switch among them. I've recently been using 5.2, but I also tried without problems last week with latest.

g-oikonomou, 2018-10-22:
I've updated all relevant wiki pages. Thanks for raising this, I'll close this (hopefully not just "for now" :))

closed 2018-10-22 by g-oikonomou