contiki-ng/contiki-ng #736: can not set the configurations on mqtt of cc26xx-web-demo, get bad request

desword, 2018-10-30:
As the title said, I try to set the broker IP as the local mqtt broker with my RPI. 
However, when I change any settings on the MQTT page, even the interval, I only get a bad request and everything is kept unchanged. 

Anyone can give me some hints?
Thanks very much.

chenek, 2018-10-30:
Try to refer to https://sunmaysky.blogspot.com/2018/04/how-to-setup-mosquitto-on-raspberry-pi.html?m=1

desword, 2018-10-30:
Thanks for your reply.
I have followed this guide, however, when I move to the step of setting the broker IP for MQTT and I still got a bad request. 

And  I already can ping6 to the cc2650 at the RPI, and the ipv6 is with aaaa::xxx.

chenek, 2018-10-30:
I think your CC26xx-web-Demo IPv6 address should be with prefix 0xfe80 not 0xaaaa. Which Contiki-NG version do you use?

desword, 2018-10-31:
I built the CC26xx-web-Demo using the examples from the contiki-os but not the Contiki-NG.

However, when I compile the example of CC26xx-web-Demo from contiki-NG, 6lbr on the RPI can not detect the sensortag running the CC26xx-web-Demo anymore.  I can not find the sensor node on the 6lbr page.

desword, 2018-10-31:
When I tried to compile the slip-radio and cc26xx-web-demo both from the Contiki-NG, with the following make command:

make TARGET=srf06-cc26xx BOARD=launchpad/cc1310 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC  slip-radio.bin"

or 

make TARGET=srf06-cc26xx BOARD=launchpad/cc1310 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC  cc26xx-web-demo.bin"

just as the guidance from:
https://sunmaysky.blogspot.com/2018/04/build-and-run-6lbrcontiki-ng-cc26xx-web.html?m=0

I encounter the following errors:

```
CC        ../../../../os/net/routing/rpl-classic/rpl-icmp6.c
../../../../os/net/routing/rpl-classic/rpl-icmp6.c: In function 'dao_output':
../../../../os/net/routing/rpl-classic/rpl-icmp6.c:1187:25: error: 'rpl_instance_t' has no member named 'dao_retransmit_timer'
     ctimer_set(&instance->dao_retransmit_timer, RPL_DAO_RETRANSMISSION_TIMEOUT,
                         ^
../../../../os/net/routing/rpl-classic/rpl-icmp6.c:1188:16: error: 'handle_dao_retransmission' undeclared (first use in this function)
                handle_dao_retransmission, parent);
                ^
../../../../os/net/routing/rpl-classic/rpl-icmp6.c:1188:16: note: each undeclared identifier is reported only once for each function it appears in
make: *** [obj_srf06-cc26xx/rpl-icmp6.o] Error 1
rm obj_srf06-cc26xx/fault-handlers.o obj_srf06-cc26xx/startup_gcc.o
```

chenek, 2018-10-31:
What is your build environment? InstantContiki 3.0 or Cygwin?

desword, 2018-10-31:
InstantContiki 3.0

chenek, 2018-10-31:
For latest Contiki-NG, you should use "make TARGET=simplelink BOARD=launchpad/cc1310 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC slip-radio.bin" instead of "make TARGET=srf06-cc26xx BOARD=launchpad/cc1310 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC slip-radio.bin""

desword, 2018-10-31:
I am sure that I have pull the latest contiki-ng, however, i got 
$ make TARGET=simplelink BOARD=launchpad/cc2650 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC slip-radio.bin
../../../../Makefile.include:84: *** The target platform "simplelink" does not exist (maybe it was misspelled?).  Stop.

No make target reply when using the TARGET=simplelink.

g-oikonomou, 2018-10-31:
If there is a problem getting something to work with Contiki-NG that is not working when it should be, please can you provide some information as to what you are trying to build, how you are trying to build it, what the error is and also the version of Contiki-NG and of the arm-gcc toolchain?

I think we may be discussing 2 or 3 different things in this thread and I admit I am somewhat confused.

g-oikonomou, 2018-10-31:
Just to add: With branch release-4.2 I tried `make TARGET=srf06-cc26xx BOARD=launchpad/cc1310 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC cc26xx-web-demo` and it worked fine. Note how I am using `cc26xx-web-demo` and not `cc26xx-web-demo.bin`

chenek, 2018-10-31:
@desword I have git latest Contiki-NG and can build slip radio with my command without problem.

desword, 2018-10-31:
I am sorry, I can now build the slip-radio with the latest contiki-ng.
I just checkout the sixlbr branch of contiki-ng, and then git pull, and I can build the slip-radio using:

make TARGET=simplelink BOARD=launchpad/cc2650 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC slip-radio.bin

In summary, to modify the MQTT broker IP, I should need:
1. The slip-radio on a cc2650 launchpad. [ok]
2. The cc26xx-web-demo on a sensortag.
3. The compiled 6lbr-development with contiki-ng. [ok]

For now, I encounter the built error for [2] as I stated in 
https://github.com/contiki-ng/contiki-ng/issues/738

Using the following command
```
make TARGET=simplelink BOARD=launchpad/cc2650 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC cc26x0-web-demo.bin
```

And I got the error:
```
CC /home/user/ng_fromcontiki/contiki-ng/arch/cpu/simplelink-cc13xx-cc26xx/rf-settings/cc26x0/ble-tx-power.c
CC ./resources/res-ble-advd.c
./resources/res-ble-advd.c:41:28: fatal error: rf-core/rf-ble.h: No such file or directory
#include "rf-core/rf-ble.h"
^
compilation terminated.
make: *** [obj_simplelink/res-ble-advd.o] Error 1
rm obj_simplelink/startup_cc13xx_cc26xx_gcc.o
```

However, there exists the file rf-ble.h after I typing find:

```
$ find . -name "rf-ble*"
./contiki-ng/arch/cpu/cc26x0-cc13x0/rf-core/rf-ble.h
```

chenek, 2018-10-31:
Try to disable BLE to build again.

desword, 2018-10-31:
I changed the "#define RF_BLE_CONF_ENABLED                   1" into "#define RF_BLE_CONF_ENABLED                   0".

However, the error is still there.

chenek, 2018-10-31:
I suppose this is caused by directory change recently in Contiki-NG. Will check later to see how to fix it.

desword, 2018-11-01:
Now, I can not access the MQTT web page.

In summary, I have all of the three components:

The slip-radio on a cc2650 launchpad. [ok]
The cc26xx-web-demo on a sensortag. [ok]- **with disabled BLE**
The compiled 6lbr-development with contiki-ng. [ok]

the contiki-ng with 6lbr is built successfully. Note the contiki is changed to contiki-ng-development.

[image]

And the following is ip address of the sensortag. I have modified the WSN address of **IP configuration** the prefix to "fd::", then I get the fd::212:4b00:c31:8d82. 
Before this modification, the address is aaaa::212:4b00:c31:8d82. 

[image]

Now I can not access mqtt web page of this sensor.

[image]

desword, 2018-11-01:
1. The slip-radio on a cc2650 launchpad. [ok]
2. The cc26xx-web-demo on a sensortag. [ok]- with disabled BLE

For the above two, I build using the following commands respectively.
```
make TARGET=simplelink BOARD=launchpad/cc2650 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC slip-radio
```

```
make TARGET=cc26x0-cc13x0 BOARD=sensortag/cc2650 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC cc26x0-web-demo
```

desword, 2018-11-01:
So, I need to checkout the tag with release/v4.0 and then rebuild and 6lbr with contiki-ng. 
That's right?

desword, 2018-11-01:
* I have pull the branch sixlbr/release 4.0, and rebuild the 6lbr+contiki-ng, It still shows `Version : 1.5.x (Contiki-NG-release/v4.0-1010-gcd3af6c3c)`, here is the git branch info for the contiki-ng.
[image]

```
git branch -v
* release/v4.0 cd3af6c Add missing part for DAO Ack test
  sixlbr       cd3af6c Add missing part for DAO Ack test
```

* I have do a factory reset of 6LBR , and the default ipv6 address has changed as expected.

[image]

* Both the border router and the PC has the route to the node to the address: `fd00::212:4b00:c31:8d82`. And they can both do ping6 to the node, however, I still can not access the MQTT config page of the node.

By the way, I compile the cc26x0-web-demo with the COAP server disable, such that I can build it for the sensortag.

g-oikonomou, 2018-11-01:
What browser are you using to access the MQTT config page? Are you running the browser on the PC? Can you open the other pages? What error are you getting while trying to access the MQTT config page?

g-oikonomou, 2018-11-02:
Did you mean all the pages on the sensortag? For the MQTT page, do you get the error when loading it? Or when trying to submit a config change? Did you try with a different browser, just in case?

desword, 2018-11-02:
No, I mean all of the pages on 6lbr configuration I can access. 
However, All of the pages related to the sensor like the **web** for config the MQTT or the COAP, I can not access.

Yes, I get the error of "ERR_CONNECTION_RESET" in Chrome, firefox and IE.

chenek, 2018-11-02:
Try to use nullrdc to see if it works.

desword, 2018-11-02:
Can you provide some links or references about how to use nullrdc?
THanks!

chenek, 2018-11-02:
Add the following lines in project-conf.h of cc26xx-web-demo example to use nullRDC to test again.

#undef NETSTACK_CONF_RDC
#define NETSTACK_CONF_RDC nullrdc_driver

g-oikonomou, 2018-11-02:
> Add the following lines in project-conf.h of cc26xx-web-demo example to use nullRDC to test again.

This is incorrect information, Contiki-NG uses a different system to switch between MAC layers and there is no such thing as RDC any more. From the command line you pasted earlier, it looks like you are using CSMA so this should be fine.

This is strange behaviour, I have not seen it before. Could you please try a different topology, without 6lbr? I would recommend trying a border router from examples/rpl-border-router and then the same web demo as you have it now.

Have a look at this guide https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-RPL-border-router

If you are building the web demo with RPL classic, you will need to make sure to also build the border router with RPL classic. Put the border router on the same device that you have been using for slip-radio. What device is it that you use?

desword, 2018-11-03:
I use the CC2650 lanuchPAD as the slip-radio, it is plugged with RPI 2. 
The RPI 2 runs the 6lbr.

I use the CC2650 sensortag as the sensor node that runs the web-demo.

desword, 2018-11-03:
Yes, I have actually tried it, however, the RPI can not ping the sensortag with the web-demo application. I plan to figure it out latter. desword 浙江大学 邮箱：desword@zju.edu.cn 签名由 网易邮箱大师 定制 On 11/03/2018 20:25, George Oikonomou wrote: You can also try putting the web demo on the sensortag and see if the web pages work

jsolderitsch, 2019-04-01:
I am having this problem now too. The simple mqtt client works but not the web demo. I also can't toggle the Tag's LED with the publish message as described in the wiki. I am using the cc1350 version of both the Tag and the Launchpad.