contiki-ng/contiki-ng #737: compile 6lbr with CONTIKI=0, fatal error: cetic-6lbr.h: No such file or directory

desword, 2018-10-31:
When I tried to compile the 6lbr on develop branch with contiki-ng on sixlbr branch, I got the error of:

  CC        ./../../../contiki-ng/os/services/ip64/ip64-arp.c
  CC        ./../../../contiki-ng/os/services/ip64/ip64-addrmap.c
  CC        ./../../../contiki-ng/os/services/ip64/ip64-ipv4-dhcp.c
./../../../contiki-ng/os/services/ip64/ip64-ipv4-dhcp.c:41:24: fatal error: cetic-6lbr.h: No such file or directory
compilation terminated.
../../../contiki-ng/Makefile.include:325: recipe for target 'obj_native/ip64-ipv4-dhcp.o' failed
make[2]: *** [obj_native/ip64-ipv4-dhcp.o] Error 1

I follow the guidance from https://github.com/cetic/6lbr/wiki/Contiki-NG

using the command:
make WITH_CONTIKI=0 all

Anyone can help me ?

laurentderu, 2018-10-31:
This should be reported on 6LBR project, not here :)

Just redo a pull on Contiki-NG sixlbr branch and it should work, you happened to pull the repositories during the small window when they where not synchronized.

Arijitdutta19910601, 2022-06-03:
> 

Could please post the command here using which you resolved the issue?