contiki-ng/contiki-ng #738: compile cc26x0-web-demo with error  fatal error: rf-core/rf-ble.h: No such file or directory

desword, 2018-10-31:
I tried to compile the cc26x0-web-demo application using the latest contiki-ng. 

with the following command:

make TARGET=simplelink BOARD=launchpad/cc2650 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC cc26x0-web-demo.bin

I got the error:

  CC        /home/user/ng_fromcontiki/contiki-ng/arch/cpu/simplelink-cc13xx-cc26xx/rf-settings/cc26x0/ble-tx-power.c
  CC        ./resources/res-ble-advd.c
./resources/res-ble-advd.c:41:28: fatal error: rf-core/rf-ble.h: No such file or directory
 #include "rf-core/rf-ble.h"
                            ^
compilation terminated.
make: *** [obj_simplelink/res-ble-advd.o] Error 1
rm obj_simplelink/startup_cc13xx_cc26xx_gcc.o

Anyone can help me ?

desword, 2018-10-31:
However, there exists the file rf-ble.h after I typing find:

$ find . -name "rf-ble*"
./contiki-ng/arch/cpu/cc26x0-cc13x0/rf-core/rf-ble.h

chenek, 2018-10-31:
Try to disable BLE to build again.

desword, 2018-10-31:
I changed the "#define RF_BLE_CONF_ENABLED 1" into "#define RF_BLE_CONF_ENABLED 0".

However, the error is still there.

chenek, 2018-10-31:
I suppose this is caused by directory change recently in Contiki-NG. Will check later to see how to fix it.

desword, 2018-10-31:
Thanks, I have noticed that although there is the file rf-ble.h in the path
```
./contiki-ng/arch/cpu/cc26x0-cc13x0/rf-core/rf-ble.h
```
However, there is no file  rf-ble.h in the path:
```
./contiki-ng/arch/cpu/simplelink/rf-core/rf-ble.h
```

Consider now that I have change the target into simplelink.

yatch, 2018-10-31:
If you're using the *latest* code on `develop` branch:

> I tried to compile the cc26x0-web-demo application using the latest contiki-ng.
>
> with the following command:
>
> make TARGET=simplelink BOARD=launchpad/cc2650 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC cc26x0-web-demo.bin

First of all, `cc26x0-web-demo` is a platform-specific example of `cc26x0-cc13x0`. So, `TARGET` should be `cc26x0-cc13x0`. Then, it should work.

```bash
$ git rev-parse --short HEAD
2b0c2d5
$ make TARGET=cc26x0-cc13x0 BOARD=launchpad/cc2650 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC cc26x0-web-demo > /dev/null
...
  CC        ../../../../arch/cpu/cc26x0-cc13x0/lib/cc26xxware/startup_files/ccfg.c
  CC        ../../../../arch/cpu/cc26x0-cc13x0/./ieee-addr.c
  CC        coap-server.c
  CC        net-uart.c
  CC        mqtt-client.c
  CC        httpd-simple.c
  CC        cetic-6lbr-client.c
  CC        ../../../../arch/cpu/cc26x0-cc13x0/./fault-handlers.c
  CC        ../../../../arch/cpu/cc26x0-cc13x0/lib/cc26xxware/startup_files/startup_gcc.c
  CC        cc26x0-web-demo.c
  LD        cc26x0-web-demo.elf
arm-none-eabi-objcopy -O ihex cc26x0-web-demo.elf cc26x0-web-demo.i16hex
srec_cat cc26x0-web-demo.i16hex -intel -o cc26x0-web-demo.hex -intel
arm-none-eabi-objcopy -O binary --gap-fill 0xff cc26x0-web-demo.elf cc26x0-web-demo.bin
cp cc26x0-web-demo.elf cc26x0-web-demo.cc26x0-cc13x0
rm obj_cc26x0-cc13x0/fault-handlers.o obj_cc26x0-cc13x0/startup_gcc.o cc26x0-web-demo.o cc26x0-web-demo.i16hex 
```

chenek, 2018-10-31:
Cool! That works! I haven’t built the example for a while and Thanks for providing solution.

desword, 2018-10-31:
THanks!
However, I got new error:

```
  LD        cc26x0-web-demo.elf
/usr/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: cc26x0-web-demo.elf section `._user_heap_stack' will not fit in region `SRAM'
/usr/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: region `SRAM' overflowed by 20 bytes
collect2: error: ld returned 1 exit status
make: *** [cc26x0-web-demo.elf] Error 1
```

chenek, 2018-10-31:
Do you build with BLE disable?

yatch, 2018-10-31:
@desword Seems you're using an old compiler. See https://github.com/contiki-ng/contiki-ng/wiki and check if you have a right tool-chain.

yatch, 2018-10-31:
I'm closing this since the issue itself is resolved.

closed 2018-10-31 by yatch

desword, 2018-11-01:
@yatch  which version of tool-chain are you using?

desword, 2018-11-01:
@yatch 
Hi, I have download the latest docker environment to build the web demo, still get the same error.
Open a new issue:
https://github.com/contiki-ng/contiki-ng/issues/739

yatch, 2018-11-01:
Here are gcc version and contiker image version:
```
$ contiker arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.2.1 20151202 (release) [ARM/embedded-5-branch revision 231848]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$ docker images contiker/contiki-ng
REPOSITORY            TAG                 IMAGE ID            CREATED             SIZE
contiker/contiki-ng   latest              604f13cd4689        4 months ago        3.33GB
```