contiki-ng/contiki-ng #739: build cc26x0-cc13x0 web-demo, got error region `SRAM' overflowed by 12 bytes

desword, 2018-11-01:
I have download the latest docker-ce compilation environment.
However, I still got the error when build cc26x0-cc13x0 web-demo using the following command:

```
make TARGET=cc26x0-cc13x0 BOARD=sensortag/cc2650 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC cc26x0-web-demo
```

```
 CC        ../../../../arch/cpu/cc26x0-cc13x0/lib/cc26xxware/startup_files/startup_gcc.c
  CC        cc26x0-web-demo.c
  CC        coap-server.c
  CC        net-uart.c
  CC        mqtt-client.c
  CC        httpd-simple.c
  CC        cetic-6lbr-client.c
  LD        cc26x0-web-demo.elf
/usr/local/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/bin/ld: cc26x0-web-demo.elf section `._user_heap_stack' will not fit in region `SRAM'
/usr/local/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/bin/ld: region `SRAM' overflowed by 12 bytes
collect2: error: ld returned 1 exit status
../../../../arch/cpu/arm/cortex-m/Makefile.cortex-m:11: recipe for target 'cc26x0-web-demo.elf' failed
make: *** [cc26x0-web-demo.elf] Error 1
rm obj_cc26x0-cc13x0/coap-server.o obj_cc26x0-cc13x0/httpd-simple.o obj_cc26x0-cc13x0/startup_gcc.o obj_cc26x0-cc13x0/net-uart.o obj_cc26x0-cc13x0/cetic-6lbr-client.o obj_cc26x0-cc13x0/fault-handlers.o cc26x0-web-demo.o obj_cc26x0-cc13x0/mqtt-client.o

```

When I using the instant Contiki 3.0 to compile, I also got the error

```
 LD        cc26x0-web-demo.elf
/usr/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: cc26x0-web-demo.elf section `._user_heap_stack' will not fit in region `SRAM'
/usr/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: region `SRAM' overflowed by 20 bytes
collect2: error: ld returned 1 exit status
make: *** [cc26x0-web-demo.elf] Error 1
```

Anyone can help me ?

desword, 2018-11-01:
update, after I disable the RF_CONF_BLE, it worked!

However, if I really want to enable the BLE communication, then what should I do.

For example, I want to conduct the BLE experiments but not the 802.15.4 based.

yatch, 2018-11-01:
> However, I still got the error when build cc26x0-cc13x0 web-demo using the following command:
> 
> make TARGET=cc26x0-cc13x0 BOARD=sensortag/cc2650 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC cc26x0-web-demo

Why do you specify `BOARD=sensortag/cc2650` this time? In Issue #738, you used `BOARD=launchpad/cc2650`. Which board is what you really want to use?

I tired again on my local machine without any modification on `project-conf.h`:

- `BOARD=launchpad/cc2650`: build success
- `BOARD=sensortag/cc2650`: build failure

```
$ git diff examples/platform-specific/cc26x0-cc13x0/cc26x0-web-demo/project-conf.h

$ grep RF_BLE_CONF_ENABLED examples/platform-specific/cc26x0-cc13x0/cc26x0-web-demo/project-conf.h
#define RF_BLE_CONF_ENABLED                   1
```

`sensortag/cc2650` needs more modules than `launchpad/cc2650`. That's why a resulting firmware becomes too big for `BOARD=sensortag/cc2650`.

desword, 2018-11-01:
I am sorry, I want to use sensortag actually. 
So I can not build the cc26x0-web-demo example on sensortage for now?

g-oikonomou, 2018-11-01:
I looked at it a bit more. It's true this example won't build off the shelf for sensortag/cc2650 and RPL classic. I'd suggest disabling the net UART and this should work. Change project-conf.h:

#define CC26XX_WEB_DEMO_CONF_NET_UART         0

desword, 2018-11-01:
Yes it works! thanks!

closed 2018-11-01 by g-oikonomou