contiki-ng/contiki-ng #740: CC1310-TSCH security not working

webver, 2018-11-01:
Build simple-node example with enabled security on cc1310 launchpad.
Node synced to the coordinator, but can't ping node from coordinator.
Can sombody confirm this?

g-oikonomou, 2018-11-01:
I am sorry, I am not sure I understand what you are trying to describe. Can you provide some more information?

* What changes did you make, if any, before building the example? Did you change e.g. `WITH_SECURITY` in `project-conf.h` or did you make any other changes? What was the command line you used for the build?
* How did you build the "co-ordinator"? What is the co-ordinator?

webver, 2018-11-02:
Set WITH_SECURITY=1, #define WITH_PERIODIC_ROUTES_PRINT 1.
For coordinator set "is_coordinator = 1" on 62 line of node.c.
No other changes. 
Enable 2 motes. 
After "association done" there are still no routing links.
- Coordinator > Routing links: 0
- Node:> Routing links: 0

There are errors in the log:
> [ERR : Frame 15.4] header ie: wrong type 8811
> [ERR : Frame 15.4] failed to parse

Full log:
[coordinator.log](https://github.com/contiki-ng/contiki-ng/files/2542373/coordinator.log)
[node.log](https://github.com/contiki-ng/contiki-ng/files/2542374/node.log)

Johan-Henry, 2019-01-09:
Hi

I have a rpl border router and a node sending UDP packets on two cc1310 launchpads with the following custom configs:
makefile:
MAKE_MAC = MAKE_MAC_TSCH
project-conf.h
#define LLSEC802154_CONF_ENABLED 1
#define LLSEC802154_CONF_USES_EXPLICIT_KEYS 1

I see the same error:
[11:24:06:783] [INFO: Frame 15.4] Out:  1 0012.4b00.0e07.c985 27 0 (27)␊
[11:24:07:621] [INFO: Frame 15.4] Out:  0 ffff.0000.0000.0000 20 19 (41)␊
[11:24:07:748] [DBG : Frame 15.4] ie type 1, current state 0␊
[11:24:07:758] [ERR : Frame 15.4] header ie: wrong type 8811␊
[11:24:09:563] [INFO: Frame 15.4] Out:  1 ffff.0000.0000.0000 20 10 (30)␊
[11:24:09:667] [DBG : Frame 15.4] ie type 0, current state 0␊
[11:24:09:677] [DBG : Frame 15.4] header ie: len 122 id 76␊
[11:24:09:677] [ERR : Frame 15.4] failed to parse

With security disabled everything works as expected, however with security enabled as above, nothing works.

Regards

hyjay, 2019-06-24:
I've been through the same problem. With CC1310, the hello-world example with TSCH works well as expected, but it fails with the link-layer security.

Can anyone help here? Thanks in advance.

akaralis, 2019-09-04:
Hi,
This problem also occurs when CC2358 is used.