contiki-ng/contiki-ng #761: TSCH with Simplelink/CC1352 on Sub Ghz/ Prop mode ?

jetacpp, 2018-11-14:
Hi guys,
Is the TSCH feature supported on CC1352 SubGhz/Prop mode yet ? The recent example 6tisch\channel-selection-demo seemed to support CC1352:

```
PLATFORMS_EXCLUDE = sky nrf52dk native
BOARDS_EXCLUDE = srf06/cc13x0 launchpad/cc1310 launchpad/cc1350 sensortag/cc2650 sensortag/cc1350
```

but after flashing, I confirmed that the prop-mode.c driver failed to init:

> [ERR : TSCH      ] ! radio does not support getting RADIO_PARAM_RX_MODE. Abort init.
> [INFO: Main      ] Starting Contiki-NG-develop/v4.2-122-g9f935f956-dirty
> [INFO: Main      ] - Routing: RPL Lite
> [INFO: Main      ] - Net: sicslowpan
> [INFO: Main      ] - MAC: TSCH
> [INFO: Main      ] - 802.15.4 PANID: 0x81a5
> [INFO: Main      ] - 802.15.4 TSCH default hopping sequence length: 4
> [INFO: Main      ] Node ID: 51126
> [INFO: Main      ] Link-layer address: 0012.4b00.18a7.c7b6
> [INFO: RPL       ] initializing
> [INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:18a7:c7b6
> [INFO: CC13xx/CC26xx] Operating frequency on Sub-1 GHz
> [INFO: CC13xx/CC26xx] RF: Channel 15[INFO: CC13xx/CC26xx]
> [INFO: CC13xx/CC26xx] Node ID: 51126
> [INFO: RPL       ] sending a DIS to ff02::1a
> [WARN: TSCH      ] ! not initialized (see earlier logs), drop outgoing packet

aalami90, 2019-06-14:
I have the same issue for the CC1352P, any news on this?
Thanks

g-oikonomou, 2019-08-31:
#1038

g-oikonomou, 2020-11-25:
#1255

closed 2020-11-25 by g-oikonomou