contiki-ng/contiki-ng #765: Standalone RPL-Lite

simonduq, 2018-11-15:
We've been contemplating making RPL-Lite a standalone RPL implementation, with its own git repository, used by Contiki-NG as a submodule. This would enable porting to other IoT OSes (Zephyr is an immediate candidate -- there is demand) or Linux for instance.

This issue is to gather thoughts and technical considerations about a standalone RPL-Lite.

mmaxus35, 2018-11-19:
It would be a great if a standalone module exits for ESP32 and ESP8266

simonduq, 2018-11-19:
@mmaxus35 do you just mean a port to these platforms? Or, why standalone?

mmaxus35, 2018-11-19:
Because, esp is very popular iot device and it is lack of mesh/rpl networking, if there is a library which is coded in c would be beneficial. Of course porting is more beneficial.

a-v-s, 2018-12-18:
Perhaps such stand-alone implementation could also be used with the lwip stack? As it already has 6LowPan over 802.15.4, but no RPL.

nikooiko, 2019-04-29:
This sounds like a great idea! I'm working on porting a new 802.15.4 radio on the zephyr OS project and I noticed that other than the Open Thread, there is no mesh support for such radios. RPL is a great alternative and provides different type of features that are useful for low power networks.

Is there any progress on this? I would be happy to assist.

simonduq, 2019-05-05:
Great to hear interest. No update yet unfortunately but will make sure to keep this thread updated when things happen! @nikooiko would you be able for instance to take care of the Zephyr port if we provide a generic API for RPL-Lite?

nikooiko, 2019-05-05:
I would be happy to do so when the time comes to it.

leoribg, 2020-09-24:
Did you create the git repository?

dgtlrift, 2025-03-22:
Can this be revisited? There hasn't been an update on this thread is some time.