contiki-ng/contiki-ng #774: Won't compile assembly files .s or .S

firmwareguru, 2018-11-24:
Compilation of assembly files with .s or .S is not working, at least when added like this, for example

`CONTIKI_TARGET_SOURCEFILES += pdm2pcm_cc26x2_gnu.s`

Results in

`process_begin: CreateProcess(NULL, as -o build/keystone/r1//obj/pdm2pcm_cc26x2_gnu.o ../../../../arch/platform/keystone/dev/pdm2pcm_cc26x2_gnu.S, ...) failed.`

(Note: using simplelink platform)

Looking through the makefiles it's pretty easy to see why.  Neither `AS` nor `ASFLAGS` are defined and the command invocation line is missing an option.

Here is what works in my case.  Note the `-c` added before `$<`.

**Makefile.include**

```
AS := $(CC)
ASFLAGS := $(CFLAGS)

....

ifndef CUSTOM_RULE_S_TO_OBJECTDIR_O
$(OBJECTDIR)/%.o: %.S | $(OBJECTDIR)
	$(TRACE_AS)
	$(Q)$(AS) $(ASFLAGS) -o $@ -c $<
$(OBJECTDIR)/%.o: %.s | $(OBJECTDIR)
	$(TRACE_AS)
	$(Q)$(AS) $(ASFLAGS) -o $@ -c $<
endif
```

https://github.com/contiki-ng/contiki-ng/issues/302
https://github.com/contiki-ng/contiki-ng/pull/321

@g-oikonomou

firmwareguru, 2018-11-25:
Sure I can do that.  What I proposed up top is a bit of a hack so I'll need you to weigh on how you think it should be resolved.

Have you thought about opening a pull request? :)

g-oikonomou, 2019-02-08:
@firmware Evan is there any progress here?

firmwareguru, 2019-02-09:
I'm working on a patch, but I can't figure out where AS is getting set in the build system.  But it is getting set to:
`AS=as`. 

I'd like to use `AS ?= $(CC)` or `ifndef AS ...` but that isn't working because of the aforementioned issue.  @g-oikonomou do you know where it is getting set?  I've searched all makefiles including arch/cpu/arm...

g-oikonomou, 2019-02-09:
I think what you are observing is GNU make setting `AS` to `as` by default since we are not setting it elsewhere. https://www.gnu.org/software/make/manual/html_node/Implicit-Variables.html

Try setting `AS` in `Makefile.arm`. Not sure whether its preferable to use `arm-none-eabi-as` or `arm-none-eabi-gcc`

closed 2019-03-16 by g-oikonomou