contiki-ng/contiki-ng #777: TSCH timeslot timing elements

g-oikonomou, 2018-11-26:
I guess this is half a question and half me moaning about inadequate documentation :)

In `os/dev/radio.h`, those are undocumented:

```C
  RADIO_CONST_TSCH_TIMING,
  RADIO_CONST_PHY_OVERHEAD,
  RADIO_CONST_BYTE_AIR_TIME,
  RADIO_CONST_DELAY_BEFORE_TX,
  RADIO_CONST_DELAY_BEFORE_RX,
  RADIO_CONST_DELAY_BEFORE_DETECT,
```

This causes confusion.

I understand that `get_value` should write to `value` a pointer to something like `tsch_timeslot_timing_us_10000`. As far as I can see, this ultimately aims to populate `tsch_timing_us` in `tsch.c`. I cannot immediately understand why this is an array with elements ordered as per `tsch_timeslot_timing_elements` instead of a struct?

g-oikonomou, 2018-11-26:
But those are all `uint16_t`, so surely we can simply have a struct and then to iterate all we need to do is use `uint16_t` pointer arithmetic. No?

simonduq, 2018-11-26:
C99 section 6.7.2.1 says "Each non-bit-field member of a structure or union object is aligned in an implementation-defined manner appropriate to its type" :p Not very helpful...

At least in practice I believe you're right, and we're already not 100% C99, so I think your proposal makes sense

g-oikonomou, 2018-11-26:
On most modern CPUs (including x86, Arm...), scalars are self-aligned. So a 2-byte type will be aligned on a 2-byte boundary. I do not remember every reading anything that would suggest it may end up being aligned on a word-size boundary. Typical struct padding is to add padding before each member such that the member's absolute address will be divisible by the member's own size.

So unless we intend to run Contiki-NG on a SPARC that came out of the 80ies... :)

closed 2019-03-03 by g-oikonomou