contiki-ng/contiki-ng #778: RF: BUF_FULL - disables the RF part on prop-mode (CC1310)

serdarkose, 2018-11-26:
The issue is related to https://github.com/contiki-os/contiki/pull/2161
----------------------------------------------------------------------------------------
# Issue 

As the above issue states, `RF: BUF_FULL` occurs due to mid or high volume RF traffic. However, it may also occur at a single node network if the node consume resources, especially at the start-up.

Once the problem occurs, its RF part can't recover

# Brief set-up details:
**HW:**
- I have a border router: RPI 2 + CC1310 launchpad
- one CC1310 board that runs the modified examples/hello-world project (code is at the end)

**Routine:**
1. Border router is up and running
2. open a new terminal on RPI and start ping6 to the node address
3. power the node..  it immediately outputs `RF:BUF_FULL` error at the start-up. After that RF part doesn't work properly. As I experience, RF part is not recovered once the error is occured.

# As I observe:
 it flows as  `cc26xx_rf_cpe1_isr()` is called after the `IRQ_RX_BUF_FULL` is assert. The ISR calls [process_poll(&rf_core_process)](https://github.com/contiki-ng/contiki-ng/blob/4d6c8898c27440413a6a7e600053bf68b699e2a6/arch/cpu/cc26x0-cc13x0/rf-core/rf-core.c#L737) as the comment above the line states, it relies on `read_frame()` will empty the buffer. My guess is that this method needs too long time to empty the buffer, meanwhile RF part is already corrupted.  I guess the RF part may already receive new packet before it frees the buffer.

Another problem I observe is that  it turns off the RF core where it was suppose to be powered on:  ` read_frame() ` calls  `rf_core_convert_rat_to_rtimer(rat_timestamp)` at [line 867](https://github.com/contiki-ng/contiki-ng/blob/4d6c8898c27440413a6a7e600053bf68b699e2a6/arch/cpu/cc26x0-cc13x0/rf-core/prop-mode.c#L867) it flows to [line 663](https://github.com/contiki-ng/contiki-ng/blob/4d6c8898c27440413a6a7e600053bf68b699e2a6/arch/cpu/cc26x0-cc13x0/rf-core/rf-core.c#L663) where it goes to `rf_is_on()` to read `PDSTAT1RFC` register's value if it is 1(RF core is ON) or 0 (Off). At the process time, the register value is **1** and ` smartrf_settings_cmd_prop_rx_adv.status` has `0x3801 ` value (RF BUFFER FULL ERR). Before returning from the function, it does verification on [line 253](https://github.com/contiki-ng/contiki-ng/blob/4d6c8898c27440413a6a7e600053bf68b699e2a6/arch/cpu/cc26x0-cc13x0/rf-core/prop-mode.c#L253) since the value is not 0x0002, then it returns `false` to the [line 272](https://github.com/contiki-ng/contiki-ng/blob/4d6c8898c27440413a6a7e600053bf68b699e2a6/arch/cpu/cc26x0-cc13x0/rf-core/prop-mode.c#L272) it causes code to think that the rf core was off earlier and it should be turned off before exiting the function. Once the RF core is turned off in that case, the RX will be never turned on again. In some cases, it turns on the RF core to transmit DIS and turns off again.  

However, it is not a pure RF core power down issue, because I manually commented out inside [off()](https://github.com/contiki-ng/contiki-ng/blob/4d6c8898c27440413a6a7e600053bf68b699e2a6/arch/cpu/cc26x0-cc13x0/rf-core/prop-mode.c#L1072)  function that prevents RF core to be disabled. However, the radio part works terrible. 

As an example due to `#define LOG_CONF_LEVEL_FRAMER 		LOG_LEVEL_DBG` printf outputs many "RF out" message on the serial output,, but wireshark/sensniff shows that only very few of the "RF out" are actually transmitted. It is the same for the "RF in" packets, it can receive very few of the actually transmitted packets by the other nodes. 

So far, I think there are two main issues to fix:
1. prevent it to false-power off, maybe remove the verification on [line 253](https://github.com/contiki-ng/contiki-ng/blob/4d6c8898c27440413a6a7e600053bf68b699e2a6/arch/cpu/cc26x0-cc13x0/rf-core/prop-mode.c#L253) or replace it as `return smartrf_settings_cmd_prop_rx_adv.status == RF_CORE_RADIO_OP_STATUS_ACTIVE || RF_CORE_RADIO_OP_STATUS_PROP_ERROR_RXBUF;` (?)
2. I guess the RF state corrupts or does no longer holds the state accurately once the RF_BUF occurs. I am not sure what is getting broken, but is it possible to immediately solve the RF: BUFF_FULL issue inside the interrupt instead of [line 737](https://github.com/contiki-ng/contiki-ng/blob/4d6c8898c27440413a6a7e600053bf68b699e2a6/arch/cpu/cc26x0-cc13x0/rf-core/rf-core.c#L737) that I think it is too slow to act,  OR add 'reset()' function for the entire RF (or wherever it has the issue) to recover it properly (?)

- Another question, once the RF_Buffer interrupt is asserted, shall the interrupt disabled till it is solved? The current implementation takes too long to clear the rf_buf and the same interrupt occurs sometime before the buffer is cleared.

Last question about the driver is that there are many `RF_CORE_CMD_ERROR` controls, but some functions are not handling the error cases more than PRINTF. Would it be better to add error handling wherever RF_CORE_CMD_ERROR may be returned? As an example:
BUF_FULL interrupt calls read_frame(), and at the end, it calls release_data_entry(), due to `rf_core_rx_is_full` variable, [it calls rx_on_prop()](https://github.com/contiki-ng/contiki-ng/blob/4d6c8898c27440413a6a7e600053bf68b699e2a6/arch/cpu/cc26x0-cc13x0/rf-core/prop-mode.c#L806) that function calls `rf_cmd_prop_rx()` and it gets error   [ if(cmd_rx_adv->status != RF_CORE_RADIO_OP_STATUS_ACTIVE) {](https://github.com/contiki-ng/contiki-ng/blob/4d6c8898c27440413a6a7e600053bf68b699e2a6/arch/cpu/cc26x0-cc13x0/rf-core/prop-mode.c#L453) then it doesn't do any error handling where it assumes that the RX is set, but since the RX couldn't set it up, then RX part doesn't work any longer, 

#  Code part
----------------------------------------------------------------------------------------------
hello-world.c
```
#include "contiki.h"
#include "project-conf.h"
#include <stdio.h> /* For printf() */

static void add_100ms_delay( void ){
      volatile unsigned long ulLoopCount = 0;
      for(ulLoopCount = 442000; ulLoopCount > 0; ulLoopCount--) 
      {
 	    watchdog_periodic();   
      }
}

/*---------------------------------------------------------------------------*/
PROCESS(hello_world_process, "Hello world process");
AUTOSTART_PROCESSES(&hello_world_process);
/*---------------------------------------------------------------------------*/
PROCESS_THREAD(hello_world_process, ev, data)
{
  static struct etimer timer;
  PROCESS_BEGIN();
  /* Setup a periodic timer that expires after 1 seconds. */
  etimer_set(&timer, CLOCK_SECOND * 1);

  while(1) {
	uint8_t i;
	add_100ms_delay();
	add_100ms_delay();             //The RF: ERROR already occurs before reaching here
	for(i=0; i <50; i++){
		printf("%"PRIu8"\t", i);
	}
[... 66 lines cut ...]
[INFO: Main      ] Starting Contiki-NG-d05faed-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xd48a
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 16262
[INFO: Main      ] Link-layer address: 0012.4b00.197b.3f86
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:197b:3f86
[DBG : CC26xx/CC13xx] With DriverLib v0.47020
[INFO: CC26xx/CC13xx] TI SmartRF06EB + CC13xx EM
[DBG : CC26xx/CC13xx] IEEE 802.15.4: No, Sub-GHz: Yes, BLE: No, Prop: No
[INFO: CC26xx/CC13xx]  RF: Channel 26
RF Error

RF: BUF_FULL

0       1       2       3       4       5       6       7       8       9       10      11  !
rf_core_send_cmd: RF was off
rf_cmd_prop_rx: send_cmd ret=0, CMDSTA=0x0000081f, status=0x0000
[INFO: Frame 15.4] In:  1 0012.4b00.13f1.e0b9 0012.4b00.197b.3f86 21 88 (109)
0       1       2       3       4       5       6       7       8       9       10      11  4
0       1       2       3       4       5       6       7       8       9       10      11  4
0       1       2       3       4       5   
```

MartinNPN, 2018-12-13:
Hello,

i had a similar problem, but can't find the reason. Your description looks plausible. The hard delay is something which should be prevented with contiki os, but sometimes it can be necessary or the program itself must be run something a longer time (flash access or something).
The RF driver should work even if the main program blocks, maybe it could loose some data but it shouldn't break. 

I will test your solution and give some feedback, but i even hope someone with more experience will look at your proposal. 

Regards,
Martin

MartinNPN, 2019-01-08:
the serial output is maybe a problem because the uart uses polling for tx, i created an issue #812 . But this would not fix this issue, cause the reason is the not proper rx buff full handling.

serdarkose, 2019-01-08:
I think that would improve the behavior as the serial output affects it as it slows down the processing time (due to hard block). but the problem still occurs if the serial output is disabled completely, maybe it has something to do with the rf buffer. I haven't understood if RF can have its own buffer and can drop the packets if there is no space instead of creating error?

lafka, 2019-04-01:
I'm a bit unclear on the purpose of `rf_is_on/0`. Should it be used to check that the device is in RX mode? the status of the last RX command? or if the RF core is powered up? From code it's definitely not clear... functions like `transmit/1` and `get_rssi/0` uses it for powering up/down the rf core; `rx_on_prop/0` seems to be using it pick the radio operation. Anyone more familiar  with the drive that has any idea?

By removing `rf_is_on/0`, keeping track of the  _intended_ state of the radio, and relying on `rf_core_is_accessible()` (or similar) to check the power state of the device the radio seems to handle the situation alot better (almost immediate recovery after a full RXBUF)

I've made a small proof of concept ( https://github.com/lafka/contiki-ng/commit/7347ff3cdac360753705136f64a90209f94fbfd1 ) which at first glance fixes the mentioned scenario.

Using `rf_core_is_accessible/0` is probably the wrong function here. I was thinking about using `PDSTAT0:RFC_ON` but that does seem to return "maybe powered down" which is not good enough. Any suggestions on which register holds state about the Rf core power would be great.

@serdarkose If rx buffer is full new packets will not be handled.

lafka, 2023-02-02:
This can be closed now that https://github.com/contiki-ng/contiki-ng/issues/1089 is merged.

arurke, 2023-09-07:
Presumably resolved by #1089

closed 2023-09-07 by arurke