contiki-ng/contiki-ng #782: nRF52832 hello-world.flash unknown rule - How to flash it?

TT-SWP13, 2018-11-29:
Hello,

i like to flash the hello-world example on my nRF52823 over USB.
Is there an other way to flash it with Contiki-**NG**?
I follow the tutorial steps from Contiki-**OS**.:

- https://github.com/contiki-os/contiki/tree/master/platform/nrf52dk

But the command "_make TARGET=nrf52dk BOARD=nrf52832 PORT=/dev/ttyACM0 hello-world.flash_"
produce this error in Contiki-NG:

```
SDK: /home/pi/nrf5x-sdk2
SoftDevice: /home/pi/nrf5x-sdk2/components/softdevice/s1xx_iot/s1xx-iot-prototype3_nrf52_softdevice.hex
make: *** Keine Regel, um „hello-world.flash“ zu erstellen.  Schluss.
```

**My commands**
1. sudo su 
2. export NRF52_SDK_ROOT=/home/pi/nrf5x-sdk2
3. export NRF52_JLINK_PATH=/home/pi/JLink_Linux_V640_arm/
4. make TARGET=nrf52dk BOARD=nrf52832 PORT=/dev/ttyACM0 clean
5. make TARGET=nrf52dk BOARD=nrf52832 PORT=/dev/ttyACM0 erase
6. make TARGET=nrf52dk BOARD=nrf52832 PORT=/dev/ttyACM0 softdevice.flash  --> ERROR
7. make TARGET=nrf52dk BOARD=nrf52832 PORT=/dev/ttyACM0 hello-world

**With Contiki-OS it works fine. How can i flash it with Contiki-NG? Thanks!**

g-oikonomou, 2018-12-01:
Thank you for reporting this. I can confirm this as a bug. #784 fixes it.

TT-SWP13, 2018-12-02:
Thank you very much! Now it works.

PR #784 (2018-12-02): Fix the nrf52dk .jlink and .flash make targets
#698 introduced some bugs to the nrf52dk build system. This pull fixes them.

* The `%.jlink` assumes a fixed location for the output hex file. This location is no longer valid, so we fix this.
* The new location of the hex file contains slashes in the path, so we change the sed delimiter to something other than the `/` character
* We include `Makefile.cm4` earlier. This allows us to use some Make variables defined therein.
* We improve rule dependencies: `%.jlink` and `%.flash`

Fixes #782
commit 944d8da2 (2018-12-02): Fix the nrf52dk .jlink and .flash make targets

Fixes #782
closed 2018-12-02 by TT-SWP13