contiki-ng/contiki-ng #789: nbr-table null pointer exception

tiamilani, 2018-12-03:
I think there is a bug on the code of nbr-table.c in the function nbr_table_add_lladdr

On this part of the function:
```
  item = item_from_index(table, index);

  /* Initialize item data and set "used" bit */
  memset(item, 0, table->item_size);
  nbr_set_bit(used_map, table, item, 1);
```

The function item_from_index could return NULL in some cases but the function memset can throw an exception if the item is NULL.
This exception makes crash the entire program.

```
extern void *memset (void *__s, int __c, size_t __n) __THROW __nonnull ((1));
```

this I think that could be a solution:

```
/* Get item in the current table */
  item = item_from_index(table, index);

  /* Initialize item data and set "used" bit */
  if(item != NULL) {
      memset(item, 0, table->item_size);
      nbr_set_bit(used_map, table, item, 1);
  }
 ...
  return item;
```

But I don't know if it's the correct way to handle this problem in this Contiki library, I don't think I have enough knowledge of the library to pretend to be right.
If you think that this is the correct way maybe I can create a pull request to update the library.

simonduq, 2018-12-04:
When reaching this line, `index` is guaranteed to not be  `-1`, so no check should be needed.
Or maybe if `table` is `NULL`? As a defensive programming measure it might be a good idea to check that `table != NULL`.

tiamilani, 2018-12-04:
> Or maybe if `table` is `NULL`? As a defensive programming measure it might be a good idea to check that `table != NULL`.

Yes, if `table` is `NULL`, `item_from_index` will return `NULL` and `memset` will produce the exception.
It could be a good idea check if `table != NULL` before calling the function to add the element or inside it.
For a defensive programming approach, I think that could be a good idea to check the `table` or check the `item` before calling the `memset` function.

simonduq, 2018-12-04:
Will add the check, thanks!

tiamilani, 2018-12-04:
You're welcome :)

closed 2019-02-26 by simonduq