contiki-ng/contiki-ng #793: Can "CONN_PREPROCESSING_TIME_TICKS " set to be dynamic value in ble-hal-cc26xx.c?

howard1985, 2018-12-11:
Hi,
   In ble-hal-cc26xx.c file, there is a Macro definition: " #define CONN_PREPROCESSING_TIME_TICKS 100   /* 1.5 ms */". I have one question, can we set the preprocessing time to be dynamic value according to different smartphone that has different sleep clock accuracy?

howard1985, 2018-12-12:
And, why set CONN_PREPROCESSING_TIME_TICKS  to be 100(1.5ms)? Thanks!

howard1985, 2018-12-14:
@spoerk Can you solve my problem?  and also the follow macro define too,
#define CONN_WINDOW_WIDENING_TICKS        30   /* appr. 0.46 ms */

joakimeriksson, 2026-10-21:
Closing as a question. The values in ble-hal-cc26xx.c are fixed, conservative margins. Making window widening depend on the master's SCA (as the BLE spec describes) would be a reasonable enhancement. For further questions, please use GitHub Discussions.

closed 2026-10-21 by joakimeriksson