contiki-ng/contiki-ng #795: Cooja doesn't seem to  respect the IEEE802154_CONF_DEFAULT_CHANNEL

m-cim, 2018-12-13:
I tried to set the RF channel by defining IEEE802154_CONF_DEFAULT_CHANNEL in the project-conf. 
Although upon startup, the nodes print out the new RF channel, they do not seem to switch on that channel. Am I going about it the wrong way?

g-oikonomou, 2019-02-08:
Trying to make sure I am following you here, what behaviour would you expect to see in the Cooja platform when it comes to channel switching?

m-cim, 2019-02-11:
I would expect that devices (e.g. an UDP client and an UDP server) operating on different channels are not able to hear each other. 
When I set the IEEE802154_CONF_DEFAULT_CHANNEL to 15 for the UDP client and the IEEE802154_CONF_DEFAULT_CHANNEL to 20 for the UDP server, both devices print out that they operate on the respecting channel. Still, in cooja, they operate both at the same default channel (so they seem to ignore the define).