contiki-ng/contiki-ng #796: slip-radio build fails for CC1350 launchpad

janeksz, 2018-12-13:
following tutorial from here: https://sunmaysky.blogspot.com/2018/04/build-and-run-6lbrcontiki-ng-cc26xx-web.html 
I have errors:
$ make TARGET=srf06-cc26xx BOARD=launchpad/cc1310                           
 ../../Makefile.include:92: *** The target platform "srf06-cc26xx" does not exist (maybe it was misspelled?).  Stop.
--
Build ok on standard Contiki

g-oikonomou, 2018-12-13:
This seems to be becoming a very popular blog post, but it was written for Contiki and not Contiki-NG, so unfortunately it is not applicable here.

Please use the [Contiki-NG wiki ](https://github.com/contiki-ng/contiki-ng/wiki) for up-to-date instructions. Don't hesitate to join gitter for development chat.

closed 2018-12-13 by g-oikonomou

chenek, 2018-12-14:
@janeksz When I write the steps in blog, it's in the beginning stage of Contiki-NG and something is changed.You have to use "make TARGET=cc26x0-cc13x0 BOARD=launchpad/cc1310" instead of "make TARGET=cc26x0-cc13x0 BOARD=launchpad/cc1310 slip-radio.bin" to avoid "make: *** No rule to make target 'slip-radio.bin'" problem.

@g-oikonomou actually the steps in the blog is for Contiki-NG instead of Contiki. I provide the fix in my steps to solve the build issue.

g-oikonomou, 2018-12-14:
Ah good to know @chenek